espressif/esp-adf #125: ESP-EYE + internal DAC?

Niek, 2019-03-12:
I have the ESP-EYE v2.1 dev board, is it possible to use the internal DAC (like in https://github.com/espressif/esp-adf/tree/master/examples/get-started/play_mp3_dac) in order to play audio? Or is the only way to hook up an external DAC like the ES8388 to the SPI port?

krzychb, 2019-03-13:
Hi @Niek,

The output of two internal DACs is available on GPIO25 and 26. One of these pins is used to interface with onboard camera and the other one with onboard microphone, see  [ESP-EYE_V2.1 Reference Design](https://www.espressif.com/sites/default/files/documentation/ESP-EYE_V2.1_Reference_Design_1.zip). It is not possible to route DAC outputs to other pins. Therefore it is not possible to use internal DACs without doing board hardware modifications.

Niek, 2019-03-13:
Thanks @krzychb! Is there any way to add an external DAC to the SPI interface in order to play audio? The ESP-EYE would be a perfect board for two-way audio + video.

jason-mao, 2019-05-22:
@Niek I think external DAC could be connect to board by I2S interface.

closed 2019-05-22 by jason-mao