espressif/esp-adf #126: 用开发板lyrat跑dueros报错怎么解决

chenguang3312, 2019-03-13:
W (69373) DUEROS: duer_callback: type:0, source:0x3f806a74 data:1, data_len:0
I (69383) DUEROS: Dueros DUER_CMD_QUIT reconnect, retry_num:64
E (133389) DUEROS: Func:retry_login_timer_cb
E (133389) DUEROS: Recv Que DUER_CMD_LOGIN
I (133389) DUEROS: duer_start, len:1

I (133242,tid:3ffbe44c) lightduer_engine.c( 220):duer_engine_start, g_handler:3F80766C, len
gth:1, profile:3F807700
W (133393) DUEROS: duer_callback: type:0, source:0x3f806a74 data:2, data_len:0
E (133253,tid:3ffbe44c) lightduer_ca.c( 645):    create the configuation failed...
I (133298,tid:3ffbe44c) lightduer_ds_log_cache.c(  45):cached tracecode: 0x12010301
I (133303,tid:3ffbe44c) lightduer_connagent.c( 160):=====> event: 1[DUER_EVT_START], status
: -18
E (133312,tid:3ffbe44c) lightduer_connagent.c( 207):start fail! status:-18
E (133318,tid:3ffbe44c) lightduer_connagent.c( 286):Action failed: event: 1, status: -18
W (133326,tid:3ffbe44c) lightduer_engine.c( 812):duer_engine_stop has stopped
I (133333,tid:3ffbe44c) lightduer_connagent.c( 160):=====> event: 5[DUER_EVT_STOP], status:
 -18
I (133342,tid:3ffbe44c) lightduer_connagent.c( 257):connect stopped, status:-18!!
E (133499) DUEROS: event: 1
E (133502) DUEROS: event: DUER_EVENT_STOPPED
W (133360,tid:3ffbe44c) lightduer_events.c(  80):[lightduer_ca] <== event end = 4018D4E0, t
imespent = 118
W (133544) DUEROS: duer_callback: type:0, source:0x3f806a74 data:1, data_len:0
I (133545) DUEROS: Dueros DUER_CMD_QUIT reconnect, retry_num:128

jason-mao, 2019-03-25:
@chenguang3312 请检查你的profile是否有效！把你申请的profile，重命名为duer_profile ，替换已下文件：https://github.com/espressif/esp-adf/blob/master/components/dueros_service/duer_profile

chenguang3312, 2019-03-27:
> @ chenguang3312请检查你的个人资料是否有效！把你申请的个人资料，重命名为duer_profile，替换已下文件：[https](https://github.com/espressif/esp-adf/blob/master/components/dueros_service/duer_profile)：[//github.com/espressif/esp-adf/blob/master/components/dueros_service/duer_profile](https://github.com/espressif/esp-adf/blob/master/components/dueros_service/duer_profile)

多谢是这个问题，官方首页给的这个文件地址是这个目录 ADF_PATH/examples/dueros/main/duer_profile，我放这个main文件夹下了，所以报错

jason-mao, 2019-03-27:
@chenguang3312 我们也发现了这个问题，文档更新被遗漏，正在修复。

jason-mao, 2019-03-27:
The DuerOS example fix on 6214eb4639e82a14711422531db2425e31fce90e

closed 2019-03-27 by jason-mao

xieshang, 2019-04-02:
> @chenguang3312 请检查你的profile是否有效！把你申请的profile，重命名为duer_profile ，替换已下文件：https://github.com/espressif/esp-adf/blob/master/components/dueros_service/duer_profile

@jason-mao 

我也遇到了这个问题，也将下载的Profile压缩包中的随便一个文件修改为“duer_profile”并覆盖了“esp-adf\components\dueros_service\duer_profile”文件。

然后执行  “make flash monitor”下载到LyraT中，连接手机AP，依然提示该问题。

>  I (31) boot: ESP-IDF v3.1.1-6-g2aa9a2118-dirty 2nd stage bootloader
I (31) boot: compile time 15:49:00
I (33) boot: Enabling RNG early entropy source...
I (37) boot: SPI Speed      : 80MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00300000
I (83) boot: End of partition table
I (87) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x6532c (414508) map
I (215) esp_image: segment 1: paddr=0x00075354 vaddr=0x3ffb0000 size=0x04020 ( 16416) load
I (220) esp_image: segment 2: paddr=0x0007937c vaddr=0x3ffb4020 size=0x00000 (     0) load
I (222) esp_image: segment 3: paddr=0x00079384 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at E:/msys32/home/xsc/esp/esp-adf/esp-idf/components/freertos/xtensa_vectors.S:1685<br>
I (231) esp_image: segment 4: paddr=0x0007978c vaddr=0x40080400 size=0x06884 ( 26756) load
I (248) esp_image: segment 5: paddr=0x00080018 vaddr=0x400d0018 size=0x12e1fc (1237500) map
0x400d0018: _flash_cache_start at ??:?<br>
I (604) esp_image: segment 6: paddr=0x001ae21c vaddr=0x40086c84 size=0x11388 ( 70536) load
0x40086c84: ieee80211_output_process at ??:?<br>
I (628) esp_image: segment 7: paddr=0x001bf5ac vaddr=0x400c0000 size=0x00000 (     0) load
I (628) esp_image: segment 8: paddr=0x001bf5b4 vaddr=0x50000000 size=0x00000 (     0) load
I (648) boot: Loaded app from partition at offset 0x10000
I (648) boot: Disabling RNG early entropy source...
I (649) spiram: SPI RAM mode: flash 80m sram 80m
I (654) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (661) cpu_start: Pro cpu up.
I (665) cpu_start: Starting app cpu, entry point is 0x40081508
0x40081508: call_start_cpu1 at E:/msys32/home/xsc/esp/esp-adf/esp-idf/components/esp32/cpu_start.c:231<br>
I (0) cpu_start: App cpu up.
I (1134) spiram: SPI SRAM memory test OK
I (1135) heap_init: Initializing. RAM available for dynamic allocation:
I (1135) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1141) heap_init: At 3FFBB658 len 000249A8 (146 KiB): DRAM
I (1147) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1154) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1160) heap_init: At 4009800C len 00007FF4 (31 KiB): IRAM
I (1166) cpu_start: Pro cpu start user code
I (1171) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (1179) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (82) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (135) DUEROS: ADF version is v1.0-39-g10a0b94-dirty
I (137) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3
I (140) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3
W (175) PERIPH_TOUCH: _touch_init
I (175) SDCARD: Trying to mount with base path=/sdcard
I (226) SDCARD: CID name SA08G!<br>
I (449) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3
E (449) gpio: gpio_install_isr_service(394): GPIO isr service already installed
E (456) DUEROS: Func:dueros_task, Line:150, Malloc failed
I (460) wifi: wifi driver task: 3ffc31c0, prio:23, stack:3584, core=0
I (466) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (469) wifi: wifi firmware version: b2c9a19
I (478) ES8388_DRIVER: init,out:02, in:00
I (482) wifi: config NVS flash: enabled
I (355,tid:3ffc3490) lightduer_session.c(  44):random = 75933
I (489) wifi: config nano formating: disabled
I (491) AUDIO_HAL: Codec mode is 3, Ctrl:1
I (499) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
[... 68 lines cut ...]
I (6242,tid:3ffc3490) lightduer_connagent.c( 160):=====> event: 5[DUER_EVT_STOP], status: -18
I (6250,tid:3ffc3490) lightduer_connagent.c( 257):connect stopped, status:-18!!
E (6365) DUEROS: event: 1
E (6369) DUEROS: event: DUER_EVENT_STOPPED
W (6374) DUEROS: duer_callback: type:0, source:0x3f806bd8 data:1, data_len:0
W (6265,tid:3ffc3490) lightduer_events.c(  80):[lightduer_ca] <== event end = 4018934C, timespent = 85
I (6418) DUEROS: Dueros DUER_CMD_QUIT reconnect, retry_num:4
E (10425) DUEROS: Func:retry_login_timer_cb
E (10425) DUEROS: Recv Que DUER_CMD_LOGIN
I (10425) DUEROS: duer_start, len:1<br>
W (10428) DUEROS: duer_callback: type:0, source:0x3f806bd8 data:2, data_len:0
I (10320,tid:3ffc3490) lightduer_engine.c( 220):duer_engine_start, g_handler:3F8077D0, length:1, profile:3F807864
E (10365,tid:3ffc3490) lightduer_ca.c( 645):    create the configuation failed...
I (10372,tid:3ffc3490) lightduer_ds_log_cache.c(  45):cached tracecode: 0x12010301
I (10379,tid:3ffc3490) lightduer_connagent.c( 160):=====> event: 1[DUER_EVT_START], status: -18
E (10387,tid:3ffc3490) lightduer_connagent.c( 207):start fail! status:-18
[... cut ...]
W (10402,tid:3ffc3490) lightduer_engine.c( 812):duer_engine_stop has stopped
I (10409,tid:3ffc3490) lightduer_connagent.c( 160):=====> event: 5[DUER_EVT_STOP], status: -18
I (10417,tid:3ffc3490) lightduer_connagent.c( 257):connect stopped, status:-18!!
E (10533) DUEROS: event: 1
E (10536) DUEROS: event: DUER_EVENT_STOPPED
W (10432,tid:3ffc3490) lightduer_events.c(  80):[lightduer_ca] <== event end = 4018934C, timespent = 113
W (10554) DUEROS: duer_callback: type:0, source:0x3f806bd8 data:1, data_len:0
I (10558) DUEROS: Dueros DUER_CMD_QUIT reconnect, retry_num:8

jason-mao, 2019-04-04:
@xiesheng `lightduer_connagent.c( 207):start fail! status:-18`, 这个错误意思是无效的profile( lightduer_types.h duer_errcode_t,  DUER_ERR_PROFILE_INVALID         = -0x0012,).请再检查你的profile