espressif/esp-adf #127: 建立双管道时候，按键外设的问题

ChiShaoJun, 2019-03-14:
我这边做了一个简单的双管道demo，想通过按键的事件进行切换。管道1是连接bt播放音乐，管道2是连接AUX_IN播放音乐，硬件为ESP32_LyraT v4.3。但是使用管道2时却触发不到按键的事件，是想请问下我代码的哪个步骤出了问题。
```
/* Play music from Bluetooth device
 
   This example code is in the Public Domain (or CC0 licensed, at your option.)
 
   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "audio_element.h"
#include "audio_pipeline.h"
#include "audio_event_iface.h"
#include "audio_common.h"
#include "i2s_stream.h"
#include "esp_peripherals.h"
#include "periph_touch.h"
#include "audio_hal.h"
#include "board.h"
#include "bluetooth_service.h"
 
static const char *TAG = "BLUETOOTH_EXAMPLE";
 
#define LYRAT_TOUCH_SET     TOUCH_PAD_NUM9
#define LYRAT_TOUCH_PLAY    TOUCH_PAD_NUM8
#define LYRAT_TOUCH_VOLUP   TOUCH_PAD_NUM7
#define LYRAT_TOUCH_VOLDWN  TOUCH_PAD_NUM4
 
//状态切换
volatile uint8_t ex_audio_status = 1;
 
void app_main(void)
{
    audio_pipeline_handle_t pipeline1, pipeline2;
    audio_element_handle_t bt_stream_reader ,i2s_stream_reader, i2s_stream_writer1,i2s_stream_writer2;
 
    //nvs flash初始化
    esp_err_t err = nvs_flash_init();
    if (err == ESP_ERR_NVS_NO_FREE_PAGES) {
        // NVS partition was truncated and needs to be erased
        // Retry nvs_flash_init
        ESP_ERROR_CHECK(nvs_flash_erase());
        err = nvs_flash_init();
    }
 
    //日志级别打印
    esp_log_level_set("*", ESP_LOG_INFO);
    esp_log_level_set(TAG, ESP_LOG_DEBUG);
 
    //1.创建蓝牙服务
    ESP_LOGI(TAG, "[ 1 ] Create Bluetooth service");
    bluetooth_service_cfg_t bt_cfg = {
        .device_name = "ESP-ADF-SPEAKER",
        .mode = BLUETOOTH_A2DP_SINK,
    };
    bluetooth_service_start(&bt_cfg);
 
    //2.启动编解码器芯片
    ESP_LOGI(TAG, "[ 2 ] Start codec chip");
    audio_hal_codec_config_t audio_hal_codec_cfg =  AUDIO_HAL_ES8388_DEFAULT();
    audio_hal_codec_cfg.i2s_iface.samples = AUDIO_HAL_44K_SAMPLES;
    //音频输入设置把麦克风改成外部音频输入
    audio_hal_codec_cfg.adc_input = AUDIO_HAL_ADC_INPUT_LINE2;
    audio_hal_handle_t hal = audio_hal_init(&audio_hal_codec_cfg, 0);
    audio_hal_ctrl_codec(hal, AUDIO_HAL_CODEC_MODE_DECODE, AUDIO_HAL_CTRL_START);
[... 197 lines cut ...]
                            ESP_LOGI(TAG, "[ * ] audio_pipeline_pause err");
                        }
                    } else if ((int) msg2.data == TOUCH_VOLDWN) {
                        ESP_LOGI(TAG, "[ * ] [Vol-] touch tap event");
                        ex_audio_status = 0;
                        if(audio_pipeline_pause(pipeline2) != ESP_OK)
                        {
                            ESP_LOGI(TAG, "[ * ] audio_pipeline_pause err");
                        }
                        break;
                    }
                }
 
                /* Stop when the last pipeline element (i2s_stream_writer in this case) receives stop event */
                if (msg2.source_type == AUDIO_ELEMENT_TYPE_ELEMENT && msg2.source == (void *) i2s_stream_writer2
                    && msg2.cmd == AEL_MSG_CMD_REPORT_STATUS && (int) msg2.data == AEL_STATUS_STATE_STOPPED) {
                    ESP_LOGW(TAG, "[ * ] Stop event received");
                    break;
                }
            }
            break;
        }
    }
}
```

ChiShaoJun, 2019-03-20:
I changed the code according to the logic of the "flexible_pipeline".But there are a few questions.
1、Audio_pipeline_pause is replaced by audio_pipeline_stop+audio_pipeline_wait_for_stop.
2、Have to re-register i2s_stream_writer.
我根据了“flexible_pipeline”例程的逻辑重新改了代码，如下。
发现几个问题
1、audio_pipeline_pause得换成audio_pipeline_stop+audio_pipeline_wait_for_stop才可用
2、得重新注册i2s_stream_writer——audio_pipeline_register(pipeline, i2s_stream_writer, "i2sw");
```
/******************************
* 文件名称：play_bt_music_example.c
* 作    者：Kyle
* 日    期：20190319
* 描    述：bt与aux_in音频切换
******************************/

/* 头文件 --------------------------------------------------------------------*/
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "audio_element.h"
#include "audio_pipeline.h"
#include "audio_event_iface.h"
#include "audio_common.h"
#include "i2s_stream.h"
#include "esp_peripherals.h"
#include "periph_button.h"
#include "audio_hal.h"
#include "board.h"
#include "bluetooth_service.h"

/* 宏定义 --------------------------------------------------------------------*/
static const char *TAG = "BLUETOOTH_EXAMPLE";

/* 变量 ----------------------------------------------------------------------*/
//状态切换
bool ex_audio_status = false;
audio_pipeline_handle_t pipeline;
audio_element_handle_t bt_stream_reader, i2s_stream_reader, i2s_stream_writer;

/* 函数定义 ------------------------------------------------------------------*/
/******************************
* 函数名称：app_main
* 作    者：Kyle
* 日    期：20190319
* 描    述：主函数
* 输入参数：无
* 返 回 值：无
******************************/
void app_main(void)
{
    //nvs flash初始化
    esp_err_t err = nvs_flash_init();
    if (err == ESP_ERR_NVS_NO_FREE_PAGES) {
        // NVS partition was truncated and needs to be erased
        // Retry nvs_flash_init
        ESP_ERROR_CHECK(nvs_flash_erase());
        err = nvs_flash_init();
    }

    esp_log_level_set("*", ESP_LOG_INFO);
    esp_log_level_set(TAG, ESP_LOG_DEBUG);

    //1.创建蓝牙服务
    ESP_LOGI(TAG, "[ 1 ] Create Bluetooth service");
    bluetooth_service_cfg_t bt_cfg = {
        .device_name = "ESP-ADF-SPEAKER",
        .mode = BLUETOOTH_A2DP_SINK,
    };
[... 132 lines cut ...]

    //在删除侦听器之前终止管道
    /* Terminate the pipeline before removing the listener */
    audio_pipeline_remove_listener(pipeline);

    //在删除侦听器之前停止所有外围设备
    /* Stop all peripherals before removing the listener */
    esp_periph_stop_all();
    audio_event_iface_remove_listener(esp_periph_get_event_iface(), evt);

    //确保在销毁event_iface之前调用audio_pipeline_remove_listener和audio_event_iface_remove_listener
    /* Make sure audio_pipeline_remove_listener & audio_event_iface_remove_listener are called before destroying event_iface */
    audio_event_iface_destroy(evt);

    //释放所有资源
    /* Release all resources */
    audio_pipeline_deinit(pipeline);
    audio_element_deinit(bt_stream_reader);
    audio_element_deinit(i2s_stream_reader);
    audio_element_deinit(i2s_stream_writer);
    esp_periph_destroy();
    bluetooth_service_destroy();
}

```

jason-mao, 2019-05-22:
@ChiShaoJun How about this issue?

ChiShaoJun, 2019-05-22:
Finally, I chose to control the es8388 register to switch

closed 2019-05-22 by jason-mao