espressif/esp-adf #131: Need help: Create java server receive audio streaming data, for pipeline_raw_http recording example

RemyEE, 2019-03-20:
The test server written on python [https://github.com/espressif/esp-adf/blob/master/examples/recorder/pipeline_raw_http/server.py](url) works perfect.
Now I am trying to create a java server, to receive the record file in Android.
What I have done now:

- Receive all request headers

- Get the number of received byte exactly same as written byte

Now I try to get chunks size before getting chunk data, as python server does. But I can't get it work. 
In the python server, right after the while loop start (where _get_chunk_size() function occurs, the first 5 bytes always be same '4' '0' '0' '\r' '\n', which mean the chunk size is 0x400.
But, when I check the first 5 bytes needed to get chunk size, they are different all the times.
What wrong did I do? Can I have some advice to make it works?
Here is my code:
```java
import java.io.IOException;
import java.io.OutputStream;
import java.net.InetSocketAddress;

import java.util.Map;
import java.util.List;

import com.sun.net.httpserver.Headers;
import com.sun.net.httpserver.HttpExchange;
import com.sun.net.httpserver.HttpHandler;
import com.sun.net.httpserver.HttpServer;

import java.io.InputStreamReader;
import java.io.InputStream;
import java.io.OutputStream;

public class Server {

    public static void main(String[] args) throws Exception {
        HttpServer server = HttpServer.create(new InetSocketAddress(8000), 0);
        server.createContext("/upload", new Handler());
        server.setExecutor(null); // creates a default executor
        server.start();
    }

    static class Handler implements HttpHandler {
        @Override
        public void handle(HttpExchange t) throws IOException {
            /* Get request header */
            Headers requestHeaders = t.getRequestHeaders();
            
            for (Map.Entry<String, List<String>> header : requestHeaders.entrySet()) {
                
                System.out.println(header.getKey() + " : " + header.getValue());
            }
            
            /* Get request body */
            InputStream stream = t.getRequestBody();

            long cnt = 0;
            
            while (true) {
                /* Get chunk size */
                // Need help to get chunk size
                
                /* Get chunk data */
                
                
                /*  */
                if(val == -1) {
                    break;
                }
                cnt++;
            }
            stream.close();
            
            /* I checked that, the number of received byte exactly same as written byte */
            System.out.printf("Received %d bytes\n", cnt);
            
            /* Send response */
            String response = "This is the response";
            t.sendResponseHeaders(200, response.length());
            OutputStream os = t.getResponseBody();
            os.write(response.getBytes());
            os.close();
        }
    }
}
```
I am thinking that may I am using a wrong format Stream. I am not a Java-programmer btw.

XuXiangJun, 2019-03-26:
The HttpServer has resolved chunked data.
You just need read data from RequestBody until get -1.

XuXiangJun, 2019-03-26:
```
InputStream is = t.getRequestBody();
ByteArrayOutputStream byteOS = new ByteArrayOutputStream();
for (int read = is.read(); read != -1; read = is.read()) {
    byteOS.write(read);
}
// All content data
byte[] requestContent = byteOS.toByteArray();

 ...
```

RemyEE, 2019-03-27:
@XuXiangJun Thanks.
I forgot to update, I figured out the problem as @XuXiangJun said.
I can get it works now. Here is my complete code.
The request header 'X-device-id' is a header I added.
```java
import java.util.Map;
import java.util.List;
import java.util.Date;
import java.text.DateFormat;
import java.text.SimpleDateFormat;

import com.sun.net.httpserver.Headers;
import com.sun.net.httpserver.HttpExchange;
import com.sun.net.httpserver.HttpHandler;
import com.sun.net.httpserver.HttpServer;

import java.net.InetSocketAddress;

import javax.sound.sampled.*;
import java.io.*;

public class Server {

    public static void main(String[] args) throws Exception {
        HttpServer server = HttpServer.create(new InetSocketAddress(8000), 0);
        server.createContext("/upload", new Handler());
        server.setExecutor(null); // creates a default executor
        server.start();
    }

    static class Handler implements HttpHandler {
        @Override
        public void handle(HttpExchange t) throws IOException {
            /* Get request header */
            Headers requestHeaders = t.getRequestHeaders();

            String sampleRate = requestHeaders.get("x-audio-sample-rates").get(0);
            String numOfChannel = requestHeaders.get("x-audio-channel").get(0);
            String audioBits = requestHeaders.get("x-audio-bits").get(0);
            String deviceID = requestHeaders.get("X-device-id").get(0);

            /* Get request body */
            InputStream stream = t.getRequestBody();

            int read;

            ByteArrayOutputStream out = new ByteArrayOutputStream();
            byte[] buff = new byte[1024*4];

            while ((read = stream.read(buff)) > 0) {
                out.write(buff, 0, read);
            }

            /* Convert request data to byte array */
             byte[] audioBytes = out.toByteArray();

            /* Write received byte array to .wav file */
            AudioFormat frmt= new AudioFormat(
                    Integer.parseInt(sampleRate)
                    ,Integer.parseInt(audioBits)
                    ,Integer.parseInt(numOfChannel)
                    ,true
                    ,false);

            AudioInputStream ais = new AudioInputStream(
                    new ByteArrayInputStream(audioBytes)
                    ,frmt
                    ,audioBytes.length);

            DateFormat dateFormat = new SimpleDateFormat("yyyyMMddHHmmss");
            Date date = new Date();

            try {
                String wavFileName = dateFormat.format(date) + "-" + deviceID + ".wav";

                AudioSystem.write(
                    ais
                    ,AudioFileFormat.Type.WAVE
                    ,new File(wavFileName));

                System.out.println("Done writing file " + wavFileName);
            }
            catch(Exception e) {
                System.out.println("File write error");
            }

            /* Dispose all stream */
            out.flush();
            stream.close();

            /* Send response */
            String response = "This is the response";
            t.sendResponseHeaders(200, response.length());
            OutputStream os = t.getResponseBody();
            os.write(response.getBytes());
            os.close();
        }
    }
}
```

closed 2019-03-27 by jason-mao