espressif/esp-adf #137: Not using GPIO 26

HumamHelfawi, 2019-03-26:
HI,
I am using esp-adf to play a mono audio file on a single speaker. I running **play_mp3_dac** example.

I have connected GPIO25 to speaker via an amplifier. However, I am using GPIO26 for totally another purpose (parallel screen data). When I run the program, the audio is played correctly. However, the screen goes out. This is due to esp-adf trying to to utilize GPIO26 also (my guessing).

How can I tell esp-adf to not use GPIO26 at all ?

jason-mao, 2019-03-27:
@HumamHelfawi If you want to play DAC with mono, please change the DAD mode by `i2s_set_dac_mode`.

HumamHelfawi, 2019-03-27:
@jason-mao  Thanks!
`i2s_set_dac_mode `can not be used when using the esp-adf pipelines. esp-adf pipeline call this function by itself. Calling it again has no effect. 

I tried to edit this line to `I2S_DAC_CHANNEL_RIGHT_EN`:
https://github.com/espressif/esp-adf/blob/93f3a64f0677e684d7e2ade2ffa6b4d453c06bed/components/audio_stream/i2s_stream.c#L298

It worked, and no sound is outputted from `GPIO26`. However, the screen stopped which means some other part of the code initialize `GPIO26 `in somehow. 

Where else in the code I should edit?

closed 2022-02-08 by jason-mao