espressif/esp-adf #138: Speaker noise after finishing the play

HumamHelfawi, 2019-03-26:
If I run the **play_mp3_dac** example and after the sound track ends, I still hear noise from the speaker which was not before exist before playing the example.

I think that not everything being stopped after the sound track is finished. What options do I have to stop this unwanted noise?

P.S. I added while(true); at the end of the example in order to catch the noise before the program terminated.

jason-mao, 2019-04-01:
@HumamHelfawi  How is the noise?

HumamHelfawi, 2019-04-01:
it is kind of continuous light noise like an old phone with corrupted  wire (I do not know how to explain it more!)
However, I managed to stop it by adding those lines after the last line of the example:
```
i2s_set_dac_mode(I2S_DAC_CHANNEL_DISABLE);
pinMode(25, OUTPUT);
digitalWrite(25, LOW);
```

closed 2019-12-11 by jason-mao