espressif/esp-adf #142: 直接从github clone下来再编译就出错

coollofty, 2019-04-06:
git clone --recursive https://github.com/espressif/esp-adf.git
cd esp-adf/examples/get-started/play_mp3
make menuconfig

按照仓库README文件里的方式这样操作，其它多余的操作一率没有，就报错了：

esp32/esp-adf/esp-idf/make/project.mk:405: *** 多个目标匹配。 停止。

coollofty, 2019-04-06:
所有的examples里的工程包括esp-idf/examples下的，全部都是这个错误

Glen-SmarthomeG4, 2019-04-08:
问题同上，如何解决？

jason-mao, 2019-04-08:
@Glen-SmarthomeG4 @coollofty  需要设置ADF_PATH 和IDF_PATH,  参考如下说明：      https://docs.espressif.com/projects/esp-adf/en/latest/get-started/index.html#setup-path-to-esp-adf

Glen-SmarthomeG4, 2019-04-09:
已经设置了ADF_PATH 和 IDF_PATH,见如下日志打印：
smartg4-glen@smartg4-glen MINGW32 ~
# printenv ADF_PATH
D:/msys32/home/smartg4-glen/esp/esp-adf

smartg4-glen@smartg4-glen MINGW32 ~
# export IDF_PATH="D:/msys32/home/smartg4-glen/esp/esp-adf/esp-idf"

smartg4-glen@smartg4-glen MINGW32 ~
# printenv IDF_PATH
D:/msys32/home/smartg4-glen/esp/esp-adf/esp-idf

smartg4-glen@smartg4-glen MINGW32 ~
# cd ~/esp/esp-adf/examples/get-started/play_mp3

smartg4-glen@smartg4-glen MINGW32 ~/esp/esp-adf/examples/get-started/play_mp3
# make help
make[1]: 进入目录“/home/smartg4-glen/esp/esp-adf/esp-idf/tools/kconfig”
/usr/bin/gcc -c  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WI                                       DECHAR=1 -DLOCALE -MD  /home/smartg4-glen/esp/esp-adf/esp-idf/tools/kconfig/mcon                                       f.c -o mconf.o
flex -L -P zconf -o zconf.lex.c /home/smartg4-glen/esp/esp-adf/esp-idf/tools/kco                                       nfig/zconf.l
bison -t -l -p zconf -o zconf.tab.c /home/smartg4-glen/esp/esp-adf/esp-idf/tools                                       /kconfig/zconf.y
sed -E "s/\\x0D$//" /home/smartg4-glen/esp/esp-adf/esp-idf/tools/kconfig/zconf.g                                       perf | gperf -t --output-file zconf.hash.c -a -C -E -g -k '1,3,$' -p -t
/usr/bin/gcc -I /home/smartg4-glen/esp/esp-adf/esp-idf/tools/kconfig -c  -I/usr/                                       include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE -MD  z                                       conf.tab.c -o zconf.tab.o
/home/smartg4-glen/esp/esp-adf/esp-idf/tools/kconfig/lxdialog/check-lxdialog.sh                                        -check /usr/bin/gcc  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSE                                       S_WIDECHAR=1 -DLOCALE -MD -lncursesw -lintl
/usr/bin/gcc -c  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WI                                       DECHAR=1 -DLOCALE -MD  /home/smartg4-glen/esp/esp-adf/esp-idf/tools/kconfig/lxdi                                       alog/checklist.c -o lxdialog/checklist.o
/usr/bin/gcc -c  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WI                                       DECHAR=1 -DLOCALE -MD  /home/smartg4-glen/esp/esp-adf/esp-idf/tools/kconfig/lxdi                                       alog/util.c -o lxdialog/util.o
/usr/bin/gcc -c  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WI                                       DECHAR=1 -DLOCALE -MD  /home/smartg4-glen/esp/esp-adf/esp-idf/tools/kconfig/lxdi                                       alog/inputbox.c -o lxdialog/inputbox.o
/usr/bin/gcc -c  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WI                                       DECHAR=1 -DLOCALE -MD  /home/smartg4-glen/esp/esp-adf/esp-idf/tools/kconfig/lxdi                                       alog/textbox.c -o lxdialog/textbox.o
/usr/bin/gcc -c  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WI                                       DECHAR=1 -DLOCALE -MD  /home/smartg4-glen/esp/esp-adf/esp-idf/tools/kconfig/lxdi                                       alog/yesno.c -o lxdialog/yesno.o
/usr/bin/gcc -c  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WI                                       DECHAR=1 -DLOCALE -MD  /home/smartg4-glen/esp/esp-adf/esp-idf/tools/kconfig/lxdi                                       alog/menubox.c -o lxdialog/menubox.o
/usr/bin/gcc -o mconf-idf mconf.o zconf.tab.o lxdialog/checklist.o lxdialog/util                                       .o lxdialog/inputbox.o lxdialog/textbox.o lxdialog/yesno.o lxdialog/menubox.o -l                                       ncursesw -lintl
/usr/bin/gcc -c  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WI                                       DECHAR=1 -DLOCALE -MD  /home/smartg4-glen/esp/esp-adf/esp-idf/tools/kconfig/conf                                       .c -o conf.o
/usr/bin/gcc -o conf-idf conf.o  zconf.tab.o -lncursesw -lintl
make[1]: 离开目录“/home/smartg4-glen/esp/esp-adf/esp-idf/tools/kconfig”
DEFCONFIG
#
# configuration written to /home/smartg4-glen/esp/esp-adf/examples/get-started/p                                       lay_mp3/sdkconfig
#
GENCONFIG
D:/msys32/home/smartg4-glen/esp/esp-adf/esp-idf/make/project.mk:405: *** 多个目                                       标匹配。 停止。

smartg4-glen@smartg4-glen MINGW32 ~/esp/esp-adf/examples/get-started/play_mp3
# make help
D:/msys32/home/smartg4-glen/esp/esp-adf/esp-idf/make/project.mk:405: *** 多个目标匹配。 停止。

smartg4-glen@smartg4-glen MINGW32 ~/esp/esp-adf/examples/get-started/play_mp3
# make menuconfig
D:/msys32/home/smartg4-glen/esp/esp-adf/esp-idf/make/project.mk:405: *** 多个目标匹配。 停止。

Glen-SmarthomeG4, 2019-04-09:
@jason-mao 请核实问题，并提供解决方法，谢谢！

ahhfzhang, 2019-04-09:
@Glen-SmarthomeG4  请先确认一下IDF是否可以正常编译，参考文档如下：
https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html

coollofty, 2019-04-09:
> @Glen-SmarthomeG4 请先确认一下IDF是否可以正常编译，参考文档如下：
> https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html

不不，不需要做这样的检查。因为我在开发的几个项目一直可以编译，直到这一天对ESP-ADF和IDF分别做了一次git pull以及git submodule update之后，再make就不行了。从一直可以编译到不可以编译，唯一的差别就是在ADF和IDF目录下做了代码更新到最新版本的操作。其它什么都没碰。所以，检查IDF什么的都是没有意义的。

现在再正式回答一下这个IDF是否可以正常编译的问题：不可以。报出来的错误同样是这个：
**esp32/esp-adf/esp-idf/make/project.mk:405: *** 多个目标匹配。 停止。**

并且，更新之后再checkout回去之前的版本，问题还是上面这样的。当然，在一个全新的路径下重新git clone一份也是上面的这个问题。现在看来，不止我一个人碰到了这样的问题。

好在我有全盘备份的习惯，我把现在用的ESP-ADF目录删除掉，再把几天前全盘备份下的ESP-ADF目录解出来（路径当然是完全相同的），再make，就一切正常。

closed 2019-04-15 by igrr