espressif/esp-adf #143: esp/esp-adf/esp-idf/make/project.mk:405: *** multiple target patterns.  Stop.

rundhall, 2019-04-07:
Hi, I tried to recreate this example: https://docs.espressif.com/projects/esp-adf/en/latest/get-started/index.html#get-started-start-project.
I worked in a Windows environment. 
When I wanted to build: `make flash` I recived this error: 
`C:/msys32/home/user/esp/esp-adf/esp-idf/make/project.mk:405: *** multiple target patterns.  Stop.`

The code in project.mk line 405:
`COMPONENT_LINKER_DEPS ?=
$(APP_ELF): $(foreach libcomp,$(COMPONENT_LIBRARIES),$(BUILD_DIR_BASE)/$(libcomp)/lib$(libcomp).a) $(COMPONENT_LINKER_DEPS) $(COMPONENT_PROJECT_VARS)
	$(summary) LD $(patsubst $(PWD)/%,%,$@)
	$(CC) $(LDFLAGS) -o $@ -Wl,-Map=$(APP_MAP)`

I found a similar issue on this topic:
https://github.com/espressif/esp-idf/issues/166
Not solved my problem.  Any idea how to solve it?  Thanks

doayee, 2019-04-09:
+1, also experiecing this. 

I think ive tracked it down to the merge in of feature/implement_wifi_service editing the component.mk file (commit 1dd8a60) causing problems in msys on windows.

doayee, 2019-04-09:
@rundhall if you roll back to the previous commit you will probably build with no problems until they fix it

jason-mao, 2019-04-10:
@rundhall @doayee Thank you for your report. We are fixing.

rundhall, 2019-04-10:
@doayee Thanks for the tip. I just went to /esp-adf/ directory and give the order to roll back to 627f1b5 commit:
`git checkout 627f1b5`
After this, It compiled without any problem. :)

xuhongv, 2019-04-13:
Have you  solved this problem  ? @doayee

xuhongv, 2019-04-13:
[image]

closed 2019-04-15 by igrr