espressif/esp-adf #144: play a error URI lead to infinity loop

coollofty, 2019-04-08:
```
I (4960) ESP_AUDIO_TASK: It's a decoder.URI:file://sdcard/sdcard

I (4960) ESP_AUDIO_TASK: 1.IN:[file],CODEC:[auto],FILTER:[NULL],OUT:[iis],RESAMPLE:[0]
I (4970) ESP_AUDIO_CTRL: Func:media_ctrl_play, Line:405, MEM Total:4052816 Bytes, Inter:99924 Bytes, Dram:69084 Bytes

I (4980) ESP_AUDIO_TASK: 2.Handles,IN:0x3f802084,CODEC:0x3f802678,FILTER:0x0,OUT:0x3f8023b4
I (4990) ESP_AUDIO_TASK: 2.1 Update all pipelines
I (5000) ESP_AUDIO_TASK: 3. Previous starting...
I (5000) AUDIO_ELEMENT: [file] AEL_MSG_CMD_RESUME,state:1
I (5010) AUDIO_ELEMENT: [auto] AEL_MSG_CMD_RESUME,state:1
I (5020) ESP_DECODER: Ready to do audio type check (line 121)
I (5020) AUDIO_ELEMENT: [iis] AEL_MSG_CMD_RESUME,state:1
I (5030) I2S_STREAM: AUDIO_STREAM_WRITER
I (5040) -VAI-: pkg_fgetsize 'sdcard/sdcard' file not exists [2]
I (5040) FATFS_STREAM: File size is 0 byte,pos:0
E (5040) FATFS_STREAM: Failed to open file file://sdcard/sdcard
E (5050) AUDIO_ELEMENT: [file] AEL_STATUS_ERROR_OPEN
E (5060) ESP_AUDIO_TASK: media_ctrl_task error:ESP_ERR_AUDIO_OPEN
I (5060) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:1, Available:5
W (5070) AUDIO_ELEMENT: IN-[auto] AEL_IO_ABORT
E (5070) AUDIO_FORMAT: failed to get head buf data(line 319)
E (5080) ESP_DECODER: Audio type detect error (line 125)
E (5090) AUDIO_ELEMENT: [auto] AEL_STATUS_ERROR_OPEN
I (5090) AMR_DECODER: Closed
I (5100) ESP_AUDIO_TASK: Send MEDIA_CTRL_EVT_PLAY, line:485
I (5100) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PLAY, status:4
I (5110) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:2, Available:5
I (5120) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_STOP,data:0x0,size:4
I (5220) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:2, Available:5
I (5230) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_STOP,data:0x0,size:4
I (5330) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:2, Available:5
I (5330) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_STOP,data:0x0,size:4
I (5440) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:2, Available:5
I (5440) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_STOP,data:0x0,size:4
I (5540) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:2, Available:5
I (5540) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_STOP,data:0x0,size:4
I (5640) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:2, Available:5
I (5640) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_STOP,data:0x0,size:4
I (5740) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:2, Available:5
I (5740) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_STOP,data:0x0,size:4
I (5840) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:2, Available:5
I (5840) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_STOP,data:0x0,size:4
I (5940) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:2, Available:5
I (5940) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_STOP,data:0x0,size:4
I (6040) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:2, Available:5
I (6040) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_STOP,data:0x0,size:4
I (6140) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:2, Available:5
I (6140) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_STOP,data:0x0,size:4
............................
```
Played a non-existent URI, the above log will print infinitely crazy, I tried to wait for more than 5 minutes, still not stopped

jason-mao, 2019-04-08:
@coollofty Thanks for report this issue. We will fixed.

jason-mao, 2019-04-11:
@coollofty How to reproduced your issue?  i have try to `play file://sdcard/sdcard` with cli example, can't got same result like you.

jason-mao, 2019-05-22:
Same with #121

closed 2019-05-22 by jason-mao