espressif/esp-adf #145: 请问接收百度的http响应解析在哪里

jinshun0225, 2019-04-08:
想解析自定义响应，没找到百度响应解析在什么地方？

jason-mao, 2019-04-09:
这个是封装在百度SDK里面，components/clouds/dueros/lightduer的库

jinshun0225, 2019-04-09:
那请问怎么添加自定义数据节点处理呢  比如添加控制电器命令  ？

mimijx, 2019-04-13:
同问

jason-mao, 2019-04-16:
@mimijx @jinshun0225 
参考如下代码：
```c
static duer_status_t voice_ctrl_point_cb(duer_context ctx, duer_msg_t *msg, duer_addr_t *addr)
{
    ESP_LOGE(TAG, "duer_app_test_control_point");

    if (msg) {
        duer_response(msg, DUER_MSG_RSP_INVALID, NULL, 0);
    } else {
        return DUER_ERR_FAILED;
    }
    if ((msg->payload == NULL) || (msg->payload_len == 0)) {
        ESP_LOGE(TAG, "duer_msg_t failed, func:%s, len:%d", msg->payload == NULL ? "NULL" : (char *)msg->payload, msg->payload_len);
        return DUER_ERR_FAILED;
    }
    baidu_json *result = baidu_json_Parse((char *)msg->payload);
    if (result == NULL) {
        ESP_LOGE(TAG, "Failed to parse payload, func:%s, %d", __func__, __LINE__);
        return DUER_ERR_FAILED;
    }

    baidu_json *value = baidu_json_GetObjectItem(result, "value");
    if (value == NULL) {
        ESP_LOGE(TAG, "Failed to parse value, func:%s, %d", __func__, __LINE__);
        return DUER_ERR_FAILED;
    }
    int index = 0;
    if (sscanf(value->valuestring, "%u", &index) == 0) {
        ESP_LOGE(TAG, "Failed to parse valuestring, func:%s, %d", __func__, __LINE__);
        return DUER_ERR_FAILED;
    }

    // Add name of identification
    if (msg->path && (strncmp((char *)msg->path, "Music_Mode", strlen("Music_Mode")) == 0)) {
        execute_switch_mode(index);
    }
    if (msg->path && (strncmp((char *)msg->path, "Light_Color", strlen("Light_Color")) == 0)) {
        execute_setting_light_color(index);
    }
    if (msg->path && (strncmp((char *)msg->path, "Control_Light", strlen("Control_Light")) == 0)) {
        execute_ctrl_light(index);
    }

    duer_u8_t *cachedToken = malloc(msg->token_len);
    memcpy(cachedToken, msg->token, msg->token_len);
    baidu_json *payload = baidu_json_CreateObject();
    baidu_json_AddStringToObject(payload, "result", "OK");
    baidu_json_AddNumberToObject(payload, "timestamp", duer_timestamp());

    duer_seperate_response((char *)cachedToken, msg->token_len, DUER_MSG_RSP_CONTENT, payload);

    free(cachedToken);
    return DUER_OK;
}

void duer_control_point_init()
{
    duer_res_t res[] = {
        {
            DUER_RES_MODE_DYNAMIC,
            DUER_RES_OP_PUT,
            "Control_Light",
            .res.f_res = voice_ctrl_point_cb
        },
        {
            DUER_RES_MODE_DYNAMIC,
            DUER_RES_OP_PUT,
            "Music_Mode",
            .res.f_res = voice_ctrl_point_cb
        },
        {
            DUER_RES_MODE_DYNAMIC,
            DUER_RES_OP_PUT,
            "Light_Color",
            .res.f_res = voice_ctrl_point_cb
        },

    };

    duer_add_resources(res, sizeof(res) / sizeof(res[0]));
}
```
1. 其中“Control_Light”，"Music_Mode"和"Light_Color"要与DuerOS **数据点标识名**对应，如下图：

[image]

2. 在 DUER_EVENT_STARTED 的地方调用duer_control_point_init

3. 另外，数据点的申请请联系DuerOS.

closed 2019-04-16 by jason-mao