espressif/esp-adf #147: Changing volume via audio_element_setinfo does not change output volume

doayee, 2019-04-10:
Hi,

We're working with ADF on a WROVER (not the LYRA board) and are trying to implement a volume change function for the project. The audio pipeline we have constructed is as follows:

**[Bluetooth]-->bt_stream_reader-->eq-->i2s_stream_writer-->[DAC]**

We have tried to change the volume via the  `audio_element_info_t` attribute of the i2s writer element we have implemented. 

The following code shows the function we are using to do this. In this example `pl` is a custom struct which contains handles for all the audio elements of our pipeline, suffice to say that `pl->i2s.i2s_stream_writer` is of type `audio_element_handle_t` and refers to the i2s writer audio element handle.

```
void SetVolume(pipeline_t *pl, uint8_t vol)
{
	/* Get a blank audio_element_info_t */
	audio_element_info_t *info = calloc(1, sizeof(audio_element_info_t));

	/* Populate it from the current stored info so as not to overwrite any attributes */
	audio_element_getinfo(pl->i2s.i2s_stream_writer, info);

	/* Set the volume to the input volume */
	info->volume = (int)vol;

	/* Set the info */
	audio_element_setinfo(pl->i2s.i2s_stream_writer, info);

	/* Report it */
	audio_element_report_info(pl->i2s.i2s_stream_writer);
}
```

Using this function for a range of values from 0-100 produces no difference to the audio output, and this behaviour is not changed by selecting the `bluetooth_stream_reader` as the element for which the volume change is applied as opposed to the `i2s_stream_writer`.

Are we using this functionality incorrectly? Or is there a more fundamental problem in the implementation of this function?

Thanks

jason-mao, 2019-04-11:
@doayee 
Hi Doayee

As your pipeline `[Bluetooth]-->bt_stream_reader-->eq-->i2s_stream_writer-->[DAC]`, change the volume need to modify raw data value.  It's should be ALC  before [DAC] . Currently  i2s_stream has not supported ALC, which is our working for.

doayee, 2019-04-13:
thanks for this info. does this mean you plan to support this eventually?

jason-mao, 2019-04-15:
Yes

closed 2019-04-22 by doayee

SinglWolf, 2022-07-25:
Is it already possible to adjust the volume level and apply an equalizer when using an external I2S PCM5102A DAC?