espressif/esp-adf #148: 播放声音时有时会卡死

coollofty, 2019-04-11:
> I (8958) AUDIO_PIPELINE: Pipeline started
> I (9078) -VAI-: FileKey: $common.lua [7e20a53e30643dbd], not found in 1 keys
> I (9108) -VAI-: beginStep: 1, 1
> I (9108) -VAI-: add audio into list: file://sdcard/T2.amr, count=1
> I (9118) -VAI-: gotoNext addAudioToPlay file://sdcard/T2.amr
> I (9118) ESP_AUDIO_CTRL: Func:media_ctrl_play, Line:301, MEM Total:3895696 Bytes, Inter:102444 Bytes, Dram:71604 Bytes
> 
> I (9128) ESP_AUDIO_CTRL: Type:1, ext:auto len:4
>         field_set: 0xb, port: 0
>         field_data[0]: off: 0, len: 4, part: file
>         field_data[1]: off: 7, len: 6, part: sdcard
>         field_data[2]: unset
>         field_data[3]: off: 13, len: 9, part: /T2.amr
>         field_data[4]: unset
>         field_data[5]: unset
>         field_data[6]: unset
> I (9158) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5
> I (9178) ESP_AUDIO_TASK: It's a decoder.URI:file://sdcard/T2.amr
> 
> I (9178) ESP_AUDIO_TASK: 1.IN:[file],CODEC:[auto],FILTER:[NULL],OUT:[iis],RESAMPLE:[0]
> I (9178) ESP_AUDIO_TASK: 2.Handles,IN:0x3f802084,CODEC:0x3f802678,FILTER:0x0,OUT:0x3f8023b4
> I (9238) ESP_AUDIO_CTRL: Func:media_ctrl_play, Line:405, MEM Total:3895700 Bytes, Inter:102444 Bytes, Dram:71604 Bytes
> 
> I (9248) ESP_AUDIO_TASK: 2.1 Update all pipelines
> I (9258) ESP_AUDIO_TASK: 3. Previous starting...
> 
状况就是最后一瞬间的声音（零点零几秒）会不停的被重复，整个设备就卡死了，其它的代码也不再运行了，除非重启。Log则是停留在上面的状态下。这个现象有时候会发生，且都是发生在两个声音切换的瞬间，即上一个刚完立即播放下一个时

jason-mao, 2019-04-11:
@coollofty 前面你有做哪些操作，复现问题的步骤是怎样的？

coollofty, 2019-04-11:
我也在琢磨和尝试，我暂时也没有能力想复现就复现，所以才来发个issue看看有没有什么参考。我再多尝试，看看有没有更多的规律

uborfly, 2019-04-18:
> I (18158) MC_PLAYER: esp_audio_state_task g_player_queue[0].pos:0
> I (18168) MC_PLAYER: esp_auido status:3,err:0
> 
> I (18168) MC_PLAYER: app_audio play
> I (18178) ESP_AUDIO_CTRL: Func:media_ctrl_play, Line:301, MEM Total:4171412 Bytes, Inter:84052 Bytes, Dram:62148 Bytes
> 
> 	field_set: 0xb, port: 0
> 	field_data[0]: off: 0, len: 4, part: file
> 	field_data[1]: off: 7, len: 6, part: sdcard
> 	field_data[2]: unset
> 	field_data[3]: off: 13, len: 9, part: /bg/2.mp3
> 	field_data[4]: unset
> 	field_data[5]: unset
> 	field_data[6]: unset
> I (18208) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5
> I (18218) ESP_AUDIO_CTRL: Func:media_ctrl_play, Line:398, MEM Total:4171412 Bytes, Inter:84052 Bytes, Dram:62148 Bytes
> 
> I (18228) ESP_AUDIO_TASK: It's a decoder.URI:file://sdcard/bg/2.mp3
> 
> I (18238) ESP_AUDIO_TASK: 1.IN:[file],CODEC:[mp3],FILTER:[NULL],OUT:[iis],RESAMPLE:[0]
> I (18258) ESP_AUDIO_TASK: 2.Handles,IN:0x3f806b1c,CODEC:0x3f8070c0,FILTER:0x0,OUT:0x3f806d54
> I (18258) ESP_AUDIO_TASK: 2.1 Update all pipelines
> I (18278) ESP_AUDIO_TASK: 3. Previous starting...
> I (18278) AUDIO_ELEMENT: [file] AEL_MSG_CMD_RESUME,state:1
> I (18288) FATFS_STREAM: path:/sdcard/bg/2.mp3
> Task watchdog got triggered. The following tasks did not reset the watchdog in time:
>  - IDLE (CPU 0)
> Tasks currently running:
> CPU 0: el-iis
> CPU 1: IDLE
> Aborting.
> abort() was called at PC 0x400d2d43 on core 0
> 0x400d2d43: task_wdt_isr at /home/ubor/esp/esp-adf/esp-idf/components/esp32/task_wdt.c:236
> 
> 
> Backtrace: 0x4009520c:0x3ffc05b0 0x40095407:0x3ffc05d0 0x400d2d43:0x3ffc05f0 0x40084cb5:0x3ffc0610 0x4017f36d:0x00000000
> 0x4009520c: invoke_abort at /home/ubor/esp/esp-adf/esp-idf/components/esp32/panic.c:649
> 
> 0x40095407: abort at /home/ubor/esp/esp-adf/esp-idf/components/esp32/panic.c:649
> 
> 0x400d2d43: task_wdt_isr at /home/ubor/esp/esp-adf/esp-idf/components/esp32/task_wdt.c:236
> 
> 0x40084cb5: _xt_lowint1 at /home/ubor/esp/esp-adf/esp-idf/components/freertos/xtensa_vectors.S:1105
> 
> 0x4017f36d: audio_event_iface_read at /home/ubor/esp/esp-adf/components/audio_pipeline/audio_event_iface.c:152

我的也有这种现象，在这之前进行了spp下载操作，在蓝牙流程走完断开蓝牙后再启动音乐播放就会有概率出现上述log的问题，而且只要我在spp下载操作过程前在后台开启了音乐播放，会影响到蓝牙的运行，触发看门狗

jason-mao, 2019-04-19:
@uborfly 你的问题看上去应该是前面的el-iis没有结束，然后再播放mp3的时候就block了

jason-mao, 2019-05-22:
Same with #121

closed 2019-05-22 by jason-mao