espressif/esp-adf #150: Old IDF

glennhuber, 2019-04-12:
When are you going to update the associated/embedded IDF. I need the latest MDF stuff along with ADF.
Thanks.

jason-mao, 2019-04-16:
@glennhuber  Which IDF version did you used ? we have plan to update IDF to 3.2.

couperyork, 2019-04-17:
+1 Be great to have the recently released v3.2 esp-idf version @jason-mao. What would be your projected estimated delivery time for that?

glennhuber, 2019-04-17:
3.2 would be great!!! Do you have an ETA on that?

HumamHelfawi, 2019-04-24:
3.2-rc has a great fix to the mbedtls memory problem (spiram). It would be great if we moved to 3.2-rc or later. The current submodule version of esp-idf is buggy if you are https client & mqtt in the same time.

jason-mao, 2019-04-24:
Hi guys

We are working on this.  Most of the features and examples should be supported on this week.

glennhuber, 2019-04-30:
Any status updates? ETA (Estimated Time of Arrival)?

jason-mao, 2019-05-05:
@glennhuber @couperyork @HumamHelfawi IDF 3.2 has been supported on 4164305c718fcd7225a86dd3665858305239f4fa.

closed 2019-05-05 by jason-mao