espressif/esp-adf #157: ESP32-LyraT在使用dueros时模组一直重启

HaijiangHuang, 2019-04-27:
以下时串口输出日志，希望可以帮忙分析下原因，谢谢
[00:14:53.642] [0;32mI (157568) SDCARD: Trying to mount with base path=/sdcard[0m
[00:14:53.711] [0;32mI (157639) SDCARD: CID name SD!
[0m
[00:14:53.989] W (157911) wifi: wifi osi_nvs_open fail ret=4353

E (157911) wifi: wifi_init_in_caller_task 708 ret=4353
ESP_ERROR_CHECK failed: esp_err_t 0x1101 (ESP_ERR_NVS_NOT_INITIALIZED) at 0x400e8c50
file: "/mnt/f/work/8266/test/esp-adf/components/wifi_service/src/wifi_service.c" line 184
func: wifi_sta_setup
expression: esp_wifi_init(&cfg)

Backtrace: 0x400920cf:0x3ffc2cc0 0x400926d8:0x3ffc2ce0 0x400e8c50:0x3ffc2d00

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:5856
ho 0 tail 12 room 4
load:0x40078000,len:7892
load:0x40080000,len:5948
entry 0x40080320
[0;32mI (30) boot: ESP-IDF v3.1.1-6-g2aa9a2118-dirty 2nd stage bootloader[0m
[0;32mI (31) boot: compile time 23:59:35[0m
[0;32mI (33) boot: Enabling RNG early entropy source...[0m
[0;32mI (37) boot: SPI Speed      : 80MHz[0m
[0;32mI (41) boot: SPI Mode       : DIO[0m
[0;32mI (45) boot: SPI Flash Size : 4MB[0m
[0;32mI (49) boot: Partition Table:[0m
[00:14:54.103] [0;32mI (53) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (75) boot:  2 factory          factory app      00 00 00010000 00300000[0m
[0;32mI (83) boot: End of partition table[0m
[0;32mI (87) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x6608c (417932) map[0m
[00:14:54.237] [0;32mI (216) esp_image: segment 1: paddr=0x000760b4 vaddr=0x3ffb0000 size=0x04028 ( 16424) load[0m
[0;32mI (221) esp_image: segment 2: paddr=0x0007a0e4 vaddr=0x3ffb4028 size=0x00000 (     0) load[0m
[0;32mI (222) esp_image: segment 3: paddr=0x0007a0ec vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (232) esp_image: segment 4: paddr=0x0007a4f4 vaddr=0x40080400 size=0x05b1c ( 23324) load[0m
[0;32mI (248) esp_image: segment 5: paddr=0x00080018 vaddr=0x400d0018 size=0x12ed8c (1240460) map[0m
[00:14:54.627] [0;32mI (606) esp_image: segment 6: paddr=0x001aedac vaddr=0x40085f1c size=0x120f0 ( 73968) load[0m
[00:14:54.652] [0;32mI (630) esp_image: segment 7: paddr=0x001c0ea4 vaddr=0x400c0000 size=0x00000 (     0) load[0m
[0;32mI (631) esp_image: segment 8: paddr=0x001c0eac vaddr=0x50000000 size=0x00000 (     0) load[0m
[00:14:54.678] [0;32mI (651) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (651) boot: Disabling RNG early entropy source...[0m
[0;32mI (652) spiram: SPI RAM mode: flash 80m sram 80m[0m
[0;32mI (657) spiram: PSRAM initialized, cache is in low/high (2-core) mode.[0m
[0;32mI (664) cpu_start: Pro cpu up.[0m
[00:14:54.711] [0;32mI (668) cpu_start: Starting app cpu, entry point is 0x40081508[0m
[0;32mI (659) cpu_start: App cpu up.[0m
[00:14:55.158] [0;32mI (1137) spiram: SPI SRAM memory test OK[0m
[0;32mI (1137) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1138) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (1144) heap_init: At 3FFBB660 len 000249A0 (146 KiB): DRAM[0m
[0;32mI (1150) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (1156) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (1163) heap_init: At 4009800C len 00007FF4 (31 KiB): IRAM[0m
[0;32mI (1169) cpu_start: Pro cpu start user code[0m
[0;32mI (1174) spiram: Adding pool of 4096K of external SPI memory to heap allocator[0m
[0;32mI (1182) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations[0m
[00:14:55.231] [0;32mI (85) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (98) DUEROS: ADF version is v1.0-68-g316f6f7-dirty[0m
[0;32mI (99) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 [0m
[0;32mI (106) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 [0m
[... 1186 lines cut ...]
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:5856
ho 0 tail 12 room 4
load:0x40078000,len:7892
load:0x40080000,len:5948
entry 0x40080320
[0;32mI (30) boot: ESP-IDF v3.1.1-6-g2aa9a2118-dirty 2nd stage bootloader[0m
[0;32mI (31) boot: compile time 23:59:35[0m
[0;32mI (33) boot: Enabling RNG early entropy source...[0m
[0;32mI (37) boot: SPI Speed      : 80MHz[0m
[0;32mI (41) boot: SPI Mode       : DIO[0m
[0;32mI (45) boot: SPI Flash Size : 4MB[0m
[0;32mI (49) boot: Partition Table:[0m
[0;32mI (53) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (75) boot:  2 factory          factory app      00 00 00010000 00300000[0m
[0;32mI (83) boot: End of partition table[0m
[0;32mI (87) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x6608c (417932) map[0m
[00:15:22.309] [0;32mI (216) esp_image: segment 1: paddr=0x000760b4 vaddr=0x3ffb0000 size=0x04028 ( 16424) load[0m
[0;32mI (221) esp_image: segment 2: paddr=0x0007a0e4 vaddr=0x3ffb4028 size=0x00000 (     0) load[0m
[0;32mI (222) esp_image: segment 3: paddr=0x0007a0ec vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (232) esp_image: segment 4: paddr=0x0007a4f4 vaddr=0x40080400 size=0x05b1c ( 23324) load[0m
[0;32mI (248) esp_image: segment 5: paddr=0x00080018 vaddr=0x400d0018 size=0x12ed8c (1240460) map[0m

HaijiangHuang, 2019-04-27:
执行make menuconfig，依次进入component config -> Wifi -> Wifi NVS Flash，禁选该选项，编译烧录后即可。

closed 2019-04-27 by HaijiangHuang

easylady, 2022-01-18:
ESP_ERROR_CHECK failed: esp_err_t 0x1101 (ESP_ERR_NVS_NOT_INITIALIZED) at 0x400e8c50
这句话说的很明白,没有初始化nvs,或者你调用初始化nvs的位置不对