espressif/esp-adf #158: es8374_set_voice_volume compilation error

maxgerhardt, 2019-04-27:
When attempting to compile `es8374.c`, it fails because the function `es8374_set_voice_volume()` does not exist and is never declared. 

https://github.com/espressif/esp-adf/blob/316f6f77b1ee249004e09d1d6506abf0ba59aa85/components/audio_hal/driver/es8374/es8374.c#L731

However, `es8374_codec_set_voice_volume` exists:

https://github.com/espressif/esp-adf/blob/316f6f77b1ee249004e09d1d6506abf0ba59aa85/components/audio_hal/driver/es8374/es8374.c#L591-L606

If this is a bug, please fix it. Otherwise ESP-ADF is not compilable.

ichenxing, 2019-05-01:
@maxgerhardt have a try such as:
```
res |= es8374_set_voice_volume(0);  → res |= es8374_codec_set_voice_volume(0);

res |= es8374_config_i2s(cfg->codec_mode, &(cfg->i2s_iface));  → res |= es8374_codec_config_i2s(cfg->codec_mode, &(cfg->i2s_iface)); 
```

maxgerhardt, 2019-05-01:
Yeah, that's the exact fix that I proposed which works.

withdra, 2019-05-05:
@maxgerhardt 
This is a mistake we introduced when we updated the version. Because the new version did not use 8374, its compilation was omitted. If you need to use 8374, you first need to include 8374 related files in component.mk file, and there are compilation errors in the current version, which can be used normally after modification according to the prompt. We have fixed this problem and will release it in a new version.

closed 2019-05-05 by maxgerhardt