espressif/esp-adf #159: i2s_stream.h not compilable

maxgerhardt, 2019-04-27:
Compiler invocation:

```
xtensa-esp32-elf-g++ -o .pioenvs\esp32dev\src\play_mp3_example.o -c -fno-rtti -fno-exceptions -std=gnu++11 -Werror=all -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-error=unused-function -Os -Wall -nostdlib -Wpointer-arith -Wno-error=unused-but-set-variable -Wno-error=unused-variable -mlongcalls -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -DWITH_POSIX -DIDF_VER=\"3.30200.190418\" -DPLATFORMIO=40000 -DARDUINO_ESP32_DEV -DCONFIG_ESP_LYRATD_MSC_V2_2_BOARD -DESP32 -DESP_PLATFORM -DF_CPU=240000000L -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DGCC_NOT_5_2_0=0 @"C:\Users\Maxi\Desktop\pio-esp-adf-example\.pioenvs\esp32dev\longcmd-1499e514700721786b841104826a67a8" src\play_mp3_example.cpp
```

With the `longcmd` being only a tons of `-I` flags for including stuff.

```
C:\Users\Maxi\.platformio\packages\toolchain-xtensa32\bin>xtensa-esp32-elf-g++.exe --version
xtensa-esp32-elf-g++.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

The compiler fails to compile `i2s_stream.h` 

https://github.com/espressif/esp-adf/blob/316f6f77b1ee249004e09d1d6506abf0ba59aa85/components/audio_stream/include/i2s_stream.h#L41-L75

The `#define` does not initialize the struct's member in the order they were declared (e.g,, `type` then `task_prio` initialized, but `type` then `i2s_config` were declared.  

The toolchain is identical to the one advertised in the documentation [here](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup-scratch.html#alternative-setup-just-download-a-toolchain). The example code is copied from [here](https://github.com/espressif/esp-adf/blob/master/examples/get-started/play_mp3/main/play_mp3_example.c). 

What compiler are you using so that ESP-ADF compiles successfully?

maxgerhardt, 2019-04-27:
FYI the ordner in which it compiles is:

```cpp
#define I2S_STREAM_CFG_DEFAULT() {                                              \
    .type = AUDIO_STREAM_WRITER,                                                \
    .i2s_config = {                                                             \
        .mode = I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_RX,                    \
        .sample_rate = 44100,                                                   \
        .bits_per_sample = 16,                                                  \
        .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,                           \
        .communication_format = I2S_COMM_FORMAT_I2S,                            \
        .intr_alloc_flags = ESP_INTR_FLAG_LEVEL2,                               \
        .dma_buf_count = 3,                                                     \
        .dma_buf_len = 300,                                                     \
        .use_apll = 1,                                                          \
        .tx_desc_auto_clear = false,                                            \
        .fixed_mclk = 0,                                                        \
    },                                                                          \
    .i2s_port = 0,                                                              \
    .out_rb_size = I2S_STREAM_RINGBUFFER_SIZE,                                  \
    .task_stack = I2S_STREAM_TASK_STACK,                                        \
    .task_core = I2S_STREAM_TASK_CORE,                                          \
    .task_prio = I2S_STREAM_TASK_PRIO,                                          \
}
```

maxgerhardt, 2019-04-27:
Ok, actually this error occurrs because I'm compiling at as C++ and not C which has stricter rules,

closed 2019-04-27 by maxgerhardt

Dario-Ciceri, 2023-06-07:
> Ok, actually this error occurs because I'm compiling at as C++ and not C which has stricter rules.
> 
> Still, it would be nice to have it corrected.

how did you fixed it? I have this error: 

`C:/Espressif/frameworks/esp-idf-v4.4.4/workspace/arduino_as_component/components/arduino/libraries/I2S/src/I2S.cpp:148:3: warning: missing initializer for member 'esp_i2s::i2s_driver_config_t::tx_desc_auto_clear' [-Wmissing-field-initializers]`

main.cpp:

//file: main.cpp
#include "Arduino.h"

void setup(){
  Serial.begin(115200);
  while(!Serial){
    ; // wait for serial port to connect
  }
}

void loop(){
    Serial.println("loop");
    delay(1000);
}