espressif/esp-adf #161: ES8374 code is unreachable

maxgerhardt, 2019-04-27:
The code inside the ES8374 can never be compiled under normal circumstances because the component.mk file never includes it. 

https://github.com/espressif/esp-adf/blob/316f6f77b1ee249004e09d1d6506abf0ba59aa85/components/audio_hal/component.mk#L10-L30

Which board used this codec? Will it be removed or the macro re-introduced for the board which uses this codec?

withdra, 2019-05-08:
This is a mistake we introduced when we updated the version. Because the new version did not use 8374, its compilation was omitted. If you need to use 8374, you first need to include 8374 related files in component.mk file, and there are compilation errors in the current version, which can be used normally after modification according to the prompt. We have fixed this problem and will release it in a new version.

closed 2019-05-22 by jason-mao

lihanchao90, 2019-06-06:
in file esp-adf\components\audio_hal\component.mk
you should add es8374 
COMPONENT_ADD_INCLUDEDIRS +=  ./driver/es8388  ./board/es8374 **./driver/es8374**
COMPONENT_SRCDIRS += ./driver/es8388 ./board/es8374 **./driver/es8374**

jason-mao, 2019-06-06:
@lihanchao90 Thank you for your report.