espressif/esp-adf #163: Disable wakeup in recorder_engine libaray?

xiruilin, 2019-05-07:
Hi,
We need VAD, but we don't use WAKEUP. 
Is there any function that prohibits WAKEUP in recorder_engine libaray?
tks!

jason-mao, 2019-05-07:
@xiruilin Wakeup is force enabled, can't disabled. We will plan to support this.

xiruilin, 2019-05-07:
Thank you for your timely reply.

jason-mao, 2019-05-13:
@xiruilin Could you provide more details about disable wakeup in recorder_engine use case.

xiruilin, 2019-05-22:
I am sorry to get back to you late.
The prohibition of wake-up words stems from the following two points:
1. When playing radio stream (m4a format), the performance decreases a lot, and the sound is intermittent.
2. Binary files are too big. If you want to include Bluetooth and OTA, 4M Flash can't accommodate it.

jason-mao, 2019-05-22:
@xiruilin I have finished support disable the WWE by configuration, please refer to 93963a35c0fd57b40bacf229c90895ec715554ab.

> When playing radio stream (m4a format), the performance decreases a lot, and the sound is intermittent.

Improved.

> Binary files are too big. If you want to include Bluetooth and OTA, 4M Flash can't accommodate it.

I'm not sure it's whether suit for you. 

 Please check and feedback.

xiruilin, 2019-05-23:
Hi, 
This update is very good.
Whether or not wake-up words are banned, the m4a format radio stream smoothly.

But the binary file is still too big. 
Can you provide an option to disable wake-up words in esp-adf-libs/Kconfig.projbuild and decide whether to call functions related to wake-up words in librecorder_engine.a based on this option?

thanks!

xiruilin, 2019-06-20:
At new esp-adf-libs (ID:ee473006), all functions have been implemented，thanks.

closed 2019-06-24 by igrr