espressif/esp-adf #164: Play SD Card Example Not Working after IDF v3.2 updated

couperyork, 2019-05-08:
Hi there,

Thanks heaps @jason-mao for being responsive and getting the IDF update in quickly, unfortunately it has since broken our ability to read and play mp3 files of the sort from a SD card (the SD_CARD_EXAMPLE in the idf repo https://github.com/espressif/esp-idf/blob/master/examples/storage/sd_card/main/sd_card_example_main.c works fine and we have tried different SD cards and ESP modules).

The error we are receiving is as follows:
```
I (0) cpu_start: App cpu up.
I (262) heap_init: Initializing. RAM available for dynamic allocation:
I (269) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (276) heap_init: At 3FFB3168 len 0002CE98 (179 KiB): DRAM
I (281) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (288) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (296) heap_init: At 4008B4D0 len 00014B30 (82 KiB): IRAM
I (300) cpu_start: Pro cpu start user code
I (318) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (320) SDCARD_MP3_EXAMPLE: [ 1 ] Mount sdcard
I (400) SDCARD_MP3_EXAMPLE: [3.0] Create audio pipeline for playback
I (400) SDCARD_MP3_EXAMPLE: [3.1] Create fatfs stream to read data from sdcard
I (400) SDCARD_MP3_EXAMPLE: [3.2] Create i2s stream to write data to codec chip
I (410) SDCARD_MP3_EXAMPLE: [3.3] Create mp3 decoder to decode mp3 file
I (420) SDCARD_MP3_EXAMPLE: [3.4] Register all elements to audio pipeline
I (420) SDCARD_MP3_EXAMPLE: [3.5] Link it together [sdcard]-->fatfs_stream-->mp3_decoder-->i2s_stream-->[codec_chip]
I (430) SDCARD_MP3_EXAMPLE: [3.6] Set up  uri (file as fatfs_stream, mp3 as mp3 decoder, and default output is i2s)
I (440) SDCARD_MP3_EXAMPLE: [ 4 ] Set up  event listener
I (450) SDCARD_MP3_EXAMPLE: [4.1] Listening event from all elements of pipeline
I (460) SDCARD_MP3_EXAMPLE: [4.2] Listening event from peripherals
I (470) SDCARD_MP3_EXAMPLE: [ 5 ] Start audio_pipeline
E (470) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x108
E (480) diskio_sdmmc: sdmmc_read_blocks failed (264)
E (490) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x108
E (490) diskio_sdmmc: sdmmc_read_blocks failed (264)
E (500) FATFS_STREAM: Failed to open file /sdcard/home.mp3
E (500) AUDIO_ELEMENT: [file] AEL_STATUS_ERROR_OPEN
W (510) AUDIO_ELEMENT: IN-[mp3] AEL_IO_ABORT
E (510) STAGEFRIGHTMP3_DECODER: failed to read audio data (line 166)
E (520) AUDIO_ELEMENT: [mp3] AEL_STATUS_ERROR_OPEN
I (530) SDCARD_MP3_EXAMPLE: [ 6 ] Listen for all pipeline events
W (550) SDCARD_MP3_EXAMPLE: [ * ] Stop event received
I (550) SDCARD_MP3_EXAMPLE: [ 7 ] Stop audio_pipeline
W (4560) AUDIO_PIPELINE: There are no listener registered
```

The card appears to mount correctly (using SPI) and this is the example program we are using.

```
static const char *TAG = "SDCARD_MP3_EXAMPLE";

void app_main(void)
{
    audio_pipeline_handle_t pipeline;
    audio_element_handle_t fatfs_stream_reader, i2s_stream_writer, mp3_decoder;

    esp_log_level_set("*", ESP_LOG_WARN);
    esp_log_level_set(TAG, ESP_LOG_INFO);

    ESP_LOGI(TAG, "[ 1 ] Mount sdcard");
    // Initialize peripherals management
    esp_periph_config_t periph_cfg = DEFAULT_ESP_PERIPH_SET_CONFIG();
    esp_periph_set_handle_t set = esp_periph_set_init(&periph_cfg);

	sdmmc_host_t sdspi_host_config = SDSPI_HOST_DEFAULT();
	sdspi_slot_config_t sdspi_slot_config = SDSPI_SLOT_CONFIG_DEFAULT();
	esp_vfs_fat_sdmmc_mount_config_t mount_config;
	sdmmc_card_t *card_handle;
	sdspi_slot_config.gpio_miso = SD_MISO_GPIO;
	sdspi_slot_config.gpio_mosi = SD_MOSI_GPIO;
	sdspi_slot_config.gpio_sck = SD_CLK_GPIO;
	sdspi_slot_config.gpio_cs = SD_CS_GPIO;

	mount_config.format_if_mount_failed = false;
	mount_config.max_files = 3;										 //Max open files
	mount_config.allocation_unit_size = 16 * 1024; //Allocation size only used when formatting

	esp_vfs_fat_sdmmc_unmount()
	esp_err_t ret = esp_vfs_fat_sdmmc_mount("/sdcard", &sdspi_host_config, &sdspi_slot_config, &mount_config, &card_handle);
	if (ret != ESP_OK)
	{
		ESP_LOGW(TAG, "Failed to initialize the card (%s).", esp_err_to_name(ret));
	}

    ESP_LOGI(TAG, "[3.0] Create audio pipeline for playback");
    audio_pipeline_cfg_t pipeline_cfg = DEFAULT_AUDIO_PIPELINE_CONFIG();
    pipeline = audio_pipeline_init(&pipeline_cfg);
    mem_assert(pipeline);

    ESP_LOGI(TAG, "[3.1] Create fatfs stream to read data from sdcard");
    fatfs_stream_cfg_t fatfs_cfg = FATFS_STREAM_CFG_DEFAULT();
    fatfs_cfg.type = AUDIO_STREAM_READER;
    fatfs_stream_reader = fatfs_stream_init(&fatfs_cfg);

    ESP_LOGI(TAG, "[3.2] Create i2s stream to write data to codec chip");
    i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT();
    i2s_cfg.type = AUDIO_STREAM_WRITER;
    i2s_stream_writer = i2s_stream_init(&i2s_cfg);

    ESP_LOGI(TAG, "[3.3] Create mp3 decoder to decode mp3 file");
    mp3_decoder_cfg_t mp3_cfg = DEFAULT_MP3_DECODER_CONFIG();
    mp3_decoder = mp3_decoder_init(&mp3_cfg);

    ESP_LOGI(TAG, "[3.4] Register all elements to audio pipeline");
    audio_pipeline_register(pipeline, fatfs_stream_reader, "file");
    audio_pipeline_register(pipeline, mp3_decoder, "mp3");
    audio_pipeline_register(pipeline, i2s_stream_writer, "i2s");

    ESP_LOGI(TAG, "[3.5] Link it together [sdcard]-->fatfs_stream-->mp3_decoder-->i2s_stream-->[codec_chip]");
    audio_pipeline_link(pipeline, (const char *[]) {"file", "mp3", "i2s"}, 3);

    ESP_LOGI(TAG, "[3.6] Set up  uri (file as fatfs_stream, mp3 as mp3 decoder, and default output is i2s)");
    audio_element_set_uri(fatfs_stream_reader, "/sdcard/home.mp3");

    ESP_LOGI(TAG, "[ 4 ] Set up  event listener");
    audio_event_iface_cfg_t evt_cfg = AUDIO_EVENT_IFACE_DEFAULT_CFG();
    audio_event_iface_handle_t evt = audio_event_iface_init(&evt_cfg);

    ESP_LOGI(TAG, "[4.1] Listening event from all elements of pipeline");
    audio_pipeline_set_listener(pipeline, evt);

    ESP_LOGI(TAG, "[4.2] Listening event from peripherals");
    audio_event_iface_set_listener(esp_periph_set_get_event_iface(set), evt);

    ESP_LOGI(TAG, "[ 5 ] Start audio_pipeline");
    audio_pipeline_run(pipeline);

    ESP_LOGI(TAG, "[ 6 ] Listen for all pipeline events");
    while (1) {
        audio_event_iface_msg_t msg;
        esp_err_t ret = audio_event_iface_listen(evt, &msg, portMAX_DELAY);
        if (ret != ESP_OK) {
            ESP_LOGE(TAG, "[ * ] Event interface error : %d", ret);
            continue;
        }

        if (msg.source_type == AUDIO_ELEMENT_TYPE_ELEMENT && msg.source == (void *) mp3_decoder
            && msg.cmd == AEL_MSG_CMD_REPORT_MUSIC_INFO) {
            audio_element_info_t music_info = {0};
            audio_element_getinfo(mp3_decoder, &music_info);

            ESP_LOGI(TAG, "[ * ] Receive music info from mp3 decoder, sample_rates=%d, bits=%d, ch=%d",
                     music_info.sample_rates, music_info.bits, music_info.channels);

            audio_element_setinfo(i2s_stream_writer, &music_info);
            i2s_stream_set_clk(i2s_stream_writer, music_info.sample_rates, music_info.bits, music_info.channels);
            continue;
        }

        /* Stop when the last pipeline element (i2s_stream_writer in this case) receives stop event */
        if (msg.source_type == AUDIO_ELEMENT_TYPE_ELEMENT && msg.source == (void *) i2s_stream_writer
            && msg.cmd == AEL_MSG_CMD_REPORT_STATUS && (int) msg.data == AEL_STATUS_STATE_STOPPED) {
            ESP_LOGW(TAG, "[ * ] Stop event received");
            break;
        }
    }

    ESP_LOGI(TAG, "[ 7 ] Stop audio_pipeline");
    audio_pipeline_terminate(pipeline);

    audio_pipeline_unregister(pipeline, fatfs_stream_reader);
    audio_pipeline_unregister(pipeline, i2s_stream_writer);
    audio_pipeline_unregister(pipeline, mp3_decoder);

    /* Terminal the pipeline before removing the listener */
    audio_pipeline_remove_listener(pipeline);

    /* Stop all periph before removing the listener */
    esp_periph_set_stop_all(set);
    audio_event_iface_remove_listener(esp_periph_set_get_event_iface(set), evt);

    /* Make sure audio_pipeline_remove_listener & audio_event_iface_remove_listener are called before destroying event_iface */
    audio_event_iface_destroy(evt);

    /* Release all resources */
    audio_pipeline_deinit(pipeline);
    audio_element_deinit(fatfs_stream_reader);
    audio_element_deinit(i2s_stream_writer);
    audio_element_deinit(mp3_decoder);
    esp_periph_set_destroy(set);
}

```

This all worked correctly in the ADF v1.0 release and since using the latest master release with the idf upgrade has stopped working.

I found this issue (https://github.com/espressif/esp-idf/issues/1419) in the IDF which seems similar and was somewhat solved but other than that it seems quite rare.

I have tried changing the `max_freq_khz` to `SDMMC_FREQ_PROBING` to slow it down, but to no avail

Any advice of what might be going wrong would be greatly appreciated. 

Cheers!

jason-mao, 2019-05-08:
@couperyork Which hardware board used? I have test the `esp-idf/examples/storage/sd_card` example with LyraT-V4.3 board. Only works fine with SDIO mode and modify `    host.flags = SDMMC_HOST_FLAG_1BIT;`.

felixcollins, 2019-05-08:
Hi, I'm Couper's colleague. We are running this on a custom WROOM-32 based board. We are using SD in **SPI** mode (opened with esp_vfs_fat_sdmmc_mount()). It was all working fine playing mp3 before the update. We have investigated further with our code. We can still open and read files from the SD card before setting up the pipeline but when we try to do fread() from within the mp3 element callback function (set with audio_element_set_read_cb()) then we get the following log output...
`I (26851) STAGEFRIGHTMP3_DECODER: MP3 opened
E (26851) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x108
E (26861) diskio_sdmmc: sdmmc_read_blocks failed (264)
W (26871) MUSIC: Read callback read 0, errno = 5
`
MUSIC tag is our logging from the callback.

It seems that something in the way the audio pipeline or mp3 element works is upsetting the SPI.

GPIO assignments
// SDCARD
#define SD_MOSI_GPIO 5
#define SD_CS_GPIO 17
#define SD_CLK_GPIO 18
#define SD_MISO_GPIO 19
// I2S GPIOs for amp
#define I2S_LRCLK_GPIO 25
#define I2S_SCK_GPIO 26
#define I2S_SDI_GPIO 27
// I2C GPIOs for amp
#define I2C_SCL_GPIO 32
#define I2C_SDA_GPIO 33

couperyork, 2019-05-08:
Hi @jason-mao it seems as if the IDF upgrade isn't at fault but if we check out commit [93f3a64f](https://github.com/espressif/esp-adf/commit/93f3a64f0677e684d7e2ade2ffa6b4d453c06bed) it seems to stop working.

withdra, 2019-05-09:
@couperyork Did you change the i2s pins when you use this example on your board? The IO5 is used by I2S by default. When initial the I2S element, the i2s driver will be installed. Maybe SPI MOSI pin has been used by I2S.

couperyork, 2019-05-15:
So we had a few different underlying issue but I think we've found it and it wasn't caused by the IDF upgrade. 

`i2s_stream.c` is now assuming that the users are using the LYRAT boards based off the Audio Hal option in the SDK CONFIG (which also gets set by default). This behaviour was introduced in this [commit](https://github.com/espressif/esp-adf/commit/1bf61a7d98f6bd6506a1cc87a72b8b1690d3b0fc)

The following in `i2s_stream.c`:
```
        i2s_pin_config_t i2s_pin_cfg = {0};
        get_i2s_pins(i2s->config.i2s_port, &i2s_pin_cfg);
        i2s_set_pin(i2s->config.i2s_port, &i2s_pin_cfg);
```
and
```
#if (CONFIG_ESP_LYRAT_V4_3_BOARD || CONFIG_ESP_LYRAT_V4_2_BOARD)
    if (i2s->config.i2s_port == 0) {
        SET_PERI_REG_BITS(PIN_CTRL, CLK_OUT1, 0, CLK_OUT1_S);
    }
    PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_CLK_OUT1);
#endif
```
Essentially break the support for boards not LYRAT, because they now get pin configs from the board.h/ what's set in the sdkconfig/Audio Hal. 

By overriding and putting the `i2s_pin_config` back into `i2s_stream.h`, setting the i2s pin config from the passed in config in `i2s_stream_init` in `i2s_stream.c` and setting the defaults for `IIS_SCLK, IIS_LCLK, IIS_DSIN and IIS_DOUT` we were able to get our board behaving as expected before the update.

mdrobisch, 2021-09-03:
I also run into this issue with the adf version 4.1 and was unable to get the i2s stream working correctly.
For everyone have the same problem here is a workarround without patching the adf`s `i2s_stream.c` .

As shown in the getting started examples in version 4.1, you have to give a correct implementation of the `board_pins_config.c` .
Here an example for using the AI Thinker A1S Audio Kit with the AC101 codec.

```
esp_err_t get_i2s_pins(i2s_port_t port, i2s_pin_config_t *i2s_config)
{
    AUDIO_NULL_CHECK(TAG, i2s_config, return ESP_FAIL);
    if (port == I2S_NUM_0)
    {
        i2s_config->bck_io_num = GPIO_NUM_27;
        i2s_config->ws_io_num = GPIO_NUM_26;
        i2s_config->data_out_num = GPIO_NUM_25;
        i2s_config->data_in_num = GPIO_NUM_35;
        ESP_LOGI(TAG, "i2s port configured!!!!");
    }
    else
    {
        memset(i2s_config, -1, sizeof(i2s_pin_config_t));
        ESP_LOGE(TAG, "i2s port %d is not supported", port);
        return ESP_FAIL;
    }

    return ESP_OK;
}
```

Hope this helps somebody.

Maybe with a clear pointout in the example desciption, this issue could be closed by then.
The suggested solution could also be applied to the actual master version.

jason-mao, 2022-02-08:
This topic has become inactive so I'm going to close the issue. Please reopen this if you have any questions or need any further assistance.

closed 2022-02-08 by jason-mao

sobanahmed21, 2022-05-19:
The information given above helped me in debugging problem that I was facing that solution is posted for others on this [link](https://github.com/espressif/esp-adf/issues/824)