espressif/esp-adf #167: esp_audio_play play uri（iis://16000:2@from.pcm/rec.wav#file），pipeline OUT is iis

DanielMengLi, 2019-05-09:
esp_audio_play，用来play record uri（iis://16000:2@from.pcm/rec.wav#file）
#file，被正确parse 但是pipeline 依然OUT到iis

已经将正确将file writers add to esp_audio，部分代码如下：
`
        // Create readers and add to esp_audio
        file_stream_cfg_t fs_reader = FILE_STREAM_CFG_DEFAULT();
        fs_reader.type              = AUDIO_STREAM_READER;
        i2s_stream_cfg_t i2s_reader = _I2S_STREAM_CFG_DEFAULT();
        i2s_reader.type             = AUDIO_STREAM_READER;
        raw_stream_cfg_t raw_reader = _RAW_STREAM_CFG_DEFAULT();
        raw_reader.type             = AUDIO_STREAM_READER;

        esp_audio_input_stream_add(_player, raw_stream_init(&raw_reader));
        esp_audio_input_stream_add(_player, file_stream_init(&fs_reader));
        esp_audio_input_stream_add(_player, i2s_stream_init(&i2s_reader));
        http_stream_cfg_t http_cfg      = _HTTP_STREAM_CFG_DEFAULT();
        http_cfg.event_handle           = _http_stream_event_handle;
        http_cfg.type                   = AUDIO_STREAM_READER;
        http_cfg.enable_playlist_parser = true;

        audio_element_handle_t http_stream_reader = http_stream_init(&http_cfg);
        esp_audio_input_stream_add(_player, http_stream_reader);

        // Create writers and add to esp_audio
        file_stream_cfg_t fs_writer = FILE_STREAM_CFG_DEFAULT();
        fs_writer.type              = AUDIO_STREAM_WRITER;

        i2s_stream_cfg_t i2s_writer = _I2S_STREAM_CFG_DEFAULT();
        i2s_writer.type             = AUDIO_STREAM_WRITER;

        raw_stream_cfg_t raw_writer = _RAW_STREAM_CFG_DEFAULT();
        raw_writer.type             = AUDIO_STREAM_WRITER;

        esp_audio_output_stream_add(_player, i2s_stream_init(&i2s_writer));
        esp_audio_output_stream_add(_player, file_stream_init(&fs_writer));
        esp_audio_output_stream_add(_player, raw_stream_init(&raw_writer));
`
log 如下：

I (2550) ESP_AUDIO_CTRL: resample_rate:0, channels:2,rate:16000
        field_set: 0x6b, port: 0
        field_data[0]: off: 0, len: 3, part: iis
        field_data[1]: off: 14, len: 8, part: from.pcm
        field_data[2]: unset
        field_data[3]: off: 22, len: 17, part: /user/rec_000.wav
        field_data[4]: unset
        field_data[5]: off: 40, len: 4, part: file
        field_data[6]: off: 6, len: 7, part: 16000:2
I (2555) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5
I (2555) ESP_AUDIO_TASK: It's an encoder.URI:iis://16000:2@from.pcm/user/rec_000.wav#file
I (2555) ESP_AUDIO_TASK: 1.IN:[iis],CODEC:[wav],FILTER:[NULL],OUT:[iis],RESAMPLE:[0]
I (2555) ESP_AUDIO_TASK: 2.Handles,IN:0x3f8081d0,CODEC:0x3f80898c,FILTER:0x0,OUT:0x3f8083f4
I (2560) ESP_AUDIO_TASK: 2.1 Update all pipelines

jason-mao, 2019-05-09:
@DanielMengLi 我们也发现了这个问题，最近的版本会修复

DanielMengLi, 2019-05-09:
好的

closed 2019-05-22 by wujiangang