espressif/esp-adf #168: pipeline_living_stream  Task watchdog got triggered

flyoo, 2019-05-10:
```

I (0) cpu_start: App cpu up.
I (1404) spiram: SPI SRAM memory test OK
I (1405) heap_init: Initializing. RAM available for dynamic allocation:
I (1405) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1411) heap_init: At 3FFB92D8 len 00026D28 (155 KiB): DRAM
I (1417) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1424) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1430) heap_init: At 4009735C len 00008CA4 (35 KiB): IRAM
I (1437) cpu_start: Pro cpu start user code
I (1442) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (121) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (122) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (233) HTTP_LIVINGSTREAM_EXAMPLE: [ 1 ] Start audio codec chip
I (234) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (251) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (251) ES8388_DRIVER: init,out:02, in:00
I (261) AUDIO_HAL: Codec mode is 2, Ctrl:1
I (264) HTTP_LIVINGSTREAM_EXAMPLE: [2.0] Create audio pipeline for playback
I (267) HTTP_LIVINGSTREAM_EXAMPLE: [2.1] Create http stream to read data
I (275) HTTP_LIVINGSTREAM_EXAMPLE: [2.2] Create i2s stream to write data to codec chip
I (284) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (290) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (300) I2S: APLL: Req RATE: 44100, real rate: 43945.238, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11249981.000, SCLK: 1406247.625000, diva: 1, divb: 0
I (311) HTTP_LIVINGSTREAM_EXAMPLE: [2.3] Create aac decoder to decode aac file
I (319) HTTP_LIVINGSTREAM_EXAMPLE: [2.4] Register all elements to audio pipeline
I (327) HTTP_LIVINGSTREAM_EXAMPLE: [2.5] Link it together http_stream-->aac_decoder-->i2s_stream-->[codec_chip]
I (339) AUDIO_PIPELINE: link el->rb, el:0x3f8004a8, tag:http, rb:0x3f800800
I (346) AUDIO_PIPELINE: link el->rb, el:0x3f8006ec, tag:aac, rb:0x3f805840
I (353) HTTP_LIVINGSTREAM_EXAMPLE: [2.6] Set up  uri (http as http_stream, aac as aac decoder, and default output is i2s)
I (365) HTTP_LIVINGSTREAM_EXAMPLE: [ 3 ] Start and wait for Wi-Fi network
E (372) gpio: gpio_install_isr_service(394): GPIO isr service already installed
I (387) wifi: wifi driver task: 3ffc9bc8, prio:23, stack:3584, core=0
I (387) wifi: wifi firmware version: 2af77cc
I (390) wifi: config NVS flash: enabled
I (394) wifi: config nano formating: disabled
I (398) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (407) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (452) wifi: Init dynamic tx buffer num: 32
I (452) wifi: Init data frame dynamic rx buffer num: 32
I (452) wifi: Init management frame dynamic rx buffer num: 32
I (455) wifi: Init management short buffer num: 32
I (460) wifi: Init static tx buffer num: 16
I (465) wifi: Init static rx buffer size: 1600
I (468) wifi: Init static rx buffer num: 10
I (472) wifi: Init dynamic rx buffer num: 32
W (477) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (646) phy: phy_version: 4008, 544f89f, Jan 24 2019, 14:54:06, 0, 2
I (660) wifi: mode : sta (30:ae:a4:60:34:ac)
I (1753) wifi: n:9 0, o:1 0, ap:255 255, sta:9 0, prof:1
I (2895) wifi: state: init -> auth (b0)
I (2901) wifi: state: auth -> assoc (0)
I (2906) wifi: state: assoc -> run (10)
I (2919) wifi: connected with ruanjianbu, channel 9
I (2920) wifi: pm start, type: 1

I (3738) event: sta ip: 192.168.0.116, mask: 255.255.255.0, gw: 192.168.0.1
I (3740) HTTP_LIVINGSTREAM_EXAMPLE: [ 4 ] Set up  event listener
I (3743) HTTP_LIVINGSTREAM_EXAMPLE: [4.1] Listening event from all elements of pipeline
I (3751) HTTP_LIVINGSTREAM_EXAMPLE: [4.2] Listening event from peripherals
I (3759) HTTP_LIVINGSTREAM_EXAMPLE: [ 5 ] Start audio_pipeline
I (3766) AUDIO_ELEMENT: [http] Element task created
I (3771) AUDIO_ELEMENT: [aac] Element task created
I (3776) AUDIO_ELEMENT: [i2s] Element task created
I (3782) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:324, MEM Total:4310976 Bytes, Inter:211968 Bytes, Dram:176000 Bytes

I (3794) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1
I (3804) AUDIO_ELEMENT: [aac] AEL_MSG_CMD_RESUME,state:1
I (3806) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1
I (3812) I2S_STREAM: AUDIO_STREAM_WRITER
I (3817) AUDIO_PIPELINE: Pipeline started
I (3888) HTTP_STREAM: total_bytes=765
I (3967) HTTP_STREAM: total_bytes=57716
I (3971) AAC_DECODER: new aduio song
I (3971) AAC_DECODER: this audio is RAW AAC
I (4083) AAC_DECODER: I2S setup: sample_rate = 44100 : channels = 2
I (4084) AAC_DECODER: M4A sampleRate = 44100, Channels = 2, frameSize = 2048
E (8956) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (8956) task_wdt:  - IDLE0 (CPU 0)
E (8956) task_wdt: Tasks currently running:
E (8956) task_wdt: CPU 0: el-aac
E (8956) task_wdt: CPU 1: IDLE1
W (12640) HTTP_STREAM: No more data,errno:0, total_bytes:57716
I (12641) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0
I (12641) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1
I (9182) HTTP_LIVINGSTREAM_EXAMPLE: [ * ] Receive music info from aac decoder, sample_rates=44100, bits=16, ch=2
I (12680) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_PAUSE
I (12684) I2S: APLL: Req RATE: 44100, real rate: 43945.238, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11249981.000, SCLK: 1406247.625000, diva: 1, divb: 0
I (12689) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:3
I (12695) I2S_STREAM: AUDIO_STREAM_WRITER
I (12704) HTTP_STREAM: total_bytes=57462
E (17680) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (17680) task_wdt:  - IDLE0 (CPU 0)
E (17680) task_wdt: Tasks currently running:
E (17680) task_wdt: CPU 0: el-aac
E (17680) task_wdt: CPU 1: IDLE1
W (21532) HTTP_STREAM: No more data,errno:0, total_bytes:57462
I (21533) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0
I (21534) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1
I (21594) HTTP_STREAM: total_bytes=57609
E (26586) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (26586) task_wdt:  - IDLE0 (CPU 0)
E (26586) task_wdt: Tasks currently running:
E (26586) task_wdt: CPU 0: el-aac
E (26586) task_wdt: CPU 1: IDLE1
W (30466) HTTP_STREAM: No more data,errno:0, total_bytes:57609
I (30467) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0
I (30468) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1
I (30546) HTTP_STREAM: total_bytes=57353
E (39155) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (39155) task_wdt:  - IDLE0 (CPU 0)
E (39155) task_wdt: Tasks currently running:
E (39155) task_wdt: CPU 0: el-aac
E (39155) task_wdt: CPU 1: IDLE1
```
使用的pipeline_living_stream例程，声音听起来有明显的卡顿，一直在触发看门狗。

jason-mao, 2019-05-10:
@flyoo 请提供一下 sdkconfig文件和直播的URL

flyoo, 2019-05-10:
AAC_STREAM_URI "http://open.ls.qingting.fm/live/274/64k.m3u8?format=aac"

#
# Automatically generated file; DO NOT EDIT.
# Espressif IoT Development Framework Configuration
#

#
# SDK tool configuration
#
CONFIG_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_PYTHON="python"
CONFIG_MAKE_WARN_UNDEFINED_VARIABLES=y

#
# Audio HAL
#
CONFIG_ESP_LYRAT_V4_3_BOARD=y
CONFIG_ESP_LYRAT_V4_2_BOARD=
CONFIG_ESP_LYRATD_MSC_V2_1_BOARD=
CONFIG_ESP_LYRATD_MSC_V2_2_BOARD=

#
# Bootloader config
#
CONFIG_LOG_BOOTLOADER_LEVEL_NONE=
CONFIG_LOG_BOOTLOADER_LEVEL_ERROR=
CONFIG_LOG_BOOTLOADER_LEVEL_WARN=
CONFIG_LOG_BOOTLOADER_LEVEL_INFO=y
CONFIG_LOG_BOOTLOADER_LEVEL_DEBUG=
CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE=
CONFIG_LOG_BOOTLOADER_LEVEL=3
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V=
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
CONFIG_BOOTLOADER_FACTORY_RESET=
CONFIG_BOOTLOADER_APP_TEST=
CONFIG_BOOTLOADER_WDT_ENABLE=y
CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE=
CONFIG_BOOTLOADER_WDT_TIME_MS=9000

#
# Security features
#
CONFIG_SECURE_SIGNED_APPS_NO_SECURE_BOOT=
CONFIG_SECURE_BOOT_ENABLED=
CONFIG_FLASH_ENCRYPTION_ENABLED=

#
# Speech Recognition Configuration
#
CONFIG_SR_MODEL_WN3_QUANT=y
CONFIG_SR_MODEL_WN4_QUANT=
CONFIG_WAKEUP_WORD_HI_LEXIN=y
CONFIG_WAKEUP_WORD_ALEXA=

#
# Serial flasher config
#
CONFIG_ESPTOOLPY_PORT="com3"
CONFIG_ESPTOOLPY_BAUD_115200B=y
CONFIG_ESPTOOLPY_BAUD_230400B=
CONFIG_ESPTOOLPY_BAUD_921600B=
CONFIG_ESPTOOLPY_BAUD_2MB=
CONFIG_ESPTOOLPY_BAUD_OTHER=
CONFIG_ESPTOOLPY_BAUD_OTHER_VAL=115200
CONFIG_ESPTOOLPY_BAUD=115200
CONFIG_ESPTOOLPY_COMPRESSED=y
CONFIG_FLASHMODE_QIO=
CONFIG_FLASHMODE_QOUT=
CONFIG_FLASHMODE_DIO=y
[... 606 lines cut ...]
CONFIG_SPIFFS_DBG=
CONFIG_SPIFFS_API_DBG=
CONFIG_SPIFFS_GC_DBG=
CONFIG_SPIFFS_CACHE_DBG=
CONFIG_SPIFFS_CHECK_DBG=
CONFIG_SPIFFS_TEST_VISUALISATION=

#
# TCP/IP Adapter
#
CONFIG_IP_LOST_TIMER_INTERVAL=120
CONFIG_TCPIP_LWIP=y

#
# Virtual file system
#
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y

#
# Wear Levelling
#
CONFIG_WL_SECTOR_SIZE_512=
CONFIG_WL_SECTOR_SIZE_4096=y
CONFIG_WL_SECTOR_SIZE=4096

jason-mao, 2019-05-10:
这几个需要设置为如下：
CONFIG_ESPTOOLPY_FLASHFREQ_80M=y
CONFIG_ESP32_DEFAULT_CPU_FREQ_240=y
CONFIG_SPIRAM_SPEED_80M=y

flyoo, 2019-05-10:
感谢回复，确实是这个问题，已解决

closed 2019-05-10 by flyoo