espressif/esp-adf #169: Every example is showing same issue

SandiaChanchal, 2019-05-11:
I'm using ESP32-A1S board and trying to run examples of ADF but I'm getting the same error for every example that is:
 C:/esp/esp-adf/components/audio_stream/http_stream.c:501:9: error: implicit declaration of function 'esp_http_client_set_redirection' [-Werror=implicit-function-declaration]
         esp_http_client_set_redirection(http->client);
Please help me in that!!!!!!!! I've attached screenshot of error
[image]

vikramdattu, 2019-05-11:
Did you forget `git submodule update --init --recursive`? Make sure you are using idf version release/v3.2.
Also please note that, adf now uses IDF from IDF_PATH if set or from `esp-adf/esp-idf` if not.

SandiaChanchal, 2019-05-12:
> Did you forget `git submodule update --init --recursive`? Make sure you are using idf version release/v3.2.
> 
> Also please note that, adf now uses IDF from IDF_PATH if set or from `esp-adf/esp-idf` if not.
> https://github.com/espressif/esp-adf/blob/master/project.mk

SandiaChanchal, 2019-05-12:
> Did you forget `git submodule update --init --recursive`? Make sure you are using idf version release/v3.2.
> 
> Also please note that, adf now uses IDF from IDF_PATH if set or from `esp-adf/esp-idf` if not.
> https://github.com/espressif/esp-adf/blob/master/project.mk

I have done this all! Still I'm getting this error.

closed 2019-05-12 by SandiaChanchal

robin-jin, 2019-05-13:
> I'm using ESP32-A1S board and trying to run examples of ADF but I'm getting the same error for every example that is:
> C:/esp/esp-adf/components/audio_stream/http_stream.c:501:9: error: implicit [...]

I met the same issue,  have you solved it?

SandiaChanchal, 2019-05-13:
Hey I resolved this issue. You need to set path of IDF inside ADF and
delete the old one. I'm attaching screenshot of the path try that
[image: image.png]