espressif/esp-adf #170: how to read record audio data  and sent to another esp32 to play it????

TCboy1, 2019-05-15:
I want to be able to collect my recorded data in real time and then send it to another board to play, but I don't know how to read data with raw_stream? And will the recorded data be used and played on another board?
My idea is:
Record:[codec chip]-->i2s_stream-->encoder-->raw-->udp
Play:udp-->raw-->decoder--i2s--[codec chip]
Is this feasible?
Could you elaborate? very thanks.

jason-mao, 2019-05-20:
@TCboy1  Your pipeline is correct. I think you can refer https://github.com/espressif/esp-adf/blob/c099d21f21697e4e4138763e1a44319972a70dd0/examples/dueros/main/dueros_app.c#L101-#L143 to create the record data pipeline.

jason-mao, 2019-08-17:
Closed due to no more information.

closed 2019-08-17 by jason-mao