espressif/esp-adf #171: AEL_IO_ABORT when playback another audio mid of the first one still playing

manoj153, 2019-05-16:
Problem is when i am playing "1.opus" and mid of somewhere before that audio audio finish playing i trigger another audio say "2.opus" and that "2.opus" will not play when it collides . I see this ERROR  : AEL_IO_ABORT

I've followed the mp3_sdcard example with just slight modification to playback from spiffs and a opus decoder. 

Code for reference: 
` audio_pipeline_register(pipeline, spiffs_reader_el, "file");
    audio_pipeline_register(pipeline, opus_decoder, "wav");
    audio_pipeline_register(pipeline, i2s_stream_writer, "i2s");
void ENTask()

{

            while(1)
            {

                if(xSemaphoreTake(EN_xBinarySemaphore, portMAX_DELAY))
                { 
                    shot_tx(PID,TID);
                
                    audio_pipeline_stop(pipeline);
                    audio_pipeline_wait_for_stop(pipeline);
                    audio_element_set_uri(spiffs_reader_el, "/spiffs/2.opus");
                    audio_pipeline_run(pipeline);
                    gpio_intr_disable(GPIO_NUM_25);
                    xSemaphoreGive(timerTask_entrigger);
                    //shot sound 
                }

                vTaskDelay(10/portTICK_PERIOD_MS);

            }

}

from there you could see my logic and method I used to play the audio.
`

jason-mao, 2019-05-20:
@manoj153 could you help to provide more information as follow? Enable the `LOG_DEFAULT_LEVEL_DEBUG` with `make menuconfig`, and
```
    esp_log_level_set("AUDIO_ELEMENT", ESP_LOG_DEBUG);
    esp_log_level_set("AUDIO_PIPELINE", ESP_LOG_DEBUG);
```

jason-mao, 2019-12-12:
Closed by no more feedback.

closed 2019-12-12 by jason-mao