espressif/esp-adf #172: BT_APPL: TX Q overflow: (IDFGH-917)

srinath88, 2019-05-16:
## Environment

- Development Kit:      [ESP32-LyraT]
- Kit version (for WroverKit/PicoKit/DevKitC): |v3]
- Module or chip used:  [ESP32-WROVER]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.1.1-6-g2aa9a2118
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [macOS]
- Power Supply:         [USB]

## Problem Description

I am trying to stream MP3 audio file from ESP32 to a BT speaker. 

There are 2 files that play on the speaker. After the first file is played, the ESP32 does some BLE commands and then plays the second mp3 file.

Randomly, when the second file is played, there is a TX Q overflow and the audio on the speaker either breaks of plays with clicks and pops. This does not happen consistently.

```
I (25000) AUDIO_PIPELINE: Pipeline started
I (25000) AUDIO_ELEMENT: [mp3_dec] AEL_MSG_CMD_RESUME,state:1
I (25010) STAGEFRIGHTMP3_DECODER: MP3 opened
I (25010) FATFS_STREAM: File size is 309428 byte,pos:0
I (25020) AUDIO_ELEMENT: [file] el opened
I (25040) AUDIO_ELEMENT: [mp3_dec] el opened
I (25060) STAGEFRIGHTMP3_DECODER: I2S setup: sample_rate = 44100 : channels = 2
I (25100) BLUETOOTH_SOURCE_EXAMPLE: [ * ] Receive music info from mp3 decoder, sample_rates=44100, bits=16, ch=2
W (31430) BT_APPL: TX Q overflow: 12/11
W (31770) BT_APPL: TX Q overflow: 12/11
W (32430) BT_APPL: TX Q overflow: 12/11
W (32730) BT_APPL: TX Q overflow: 12/11
W (32750) BT_APPL: TX Q overflow: 12/11
W (33070) BT_APPL: TX Q overflow: 12/11
W (33090) BT_APPL: TX Q overflow: 12/11
W (33910) BT_APPL: TX Q overflow: 12/11
W (34050) BT_APPL: TX Q overflow: 12/11
W (34070) BT_APPL: TX Q overflow: 12/11
W (35230) BT_APPL: TX Q overflow: 12/11
W (35250) BT_APPL: TX Q overflow: 12/11
W (35390) BT_APPL: TX Q overflow: 12/11
```

It never happens when playing the first file. Both the files have the same sampling rate of 44.1 kHz at 16 bits.

I am stopping the pipeline, changing the uri and starting the pipeline as follows.

```
audio_pipeline_stop(pipeline_sd);
audio_pipeline_wait_for_stop(pipeline_sd);
audio_element_set_uri(fatfs_stream_reader, "/sdcard/setup.mp3");
audio_pipeline_run(pipeline_sd);
```

Is this the right way to do it? Or is the error happening because of not setting things up properly?

Thanks!

esp-xiaodong, 2019-05-20:
@srinath88  Hi，which sample project did u use? I want to reproduce it..

DarvosBR, 2019-06-23:
Did you found the solution?

closed 2025-02-18 by jason-mao