espressif/esp-adf #173: dueros duer_event_hook函数bug

ichenxing, 2019-05-23:
```
static void duer_event_hook(duer_event_t *event)
{
    if (!event) {
        ESP_LOGE(TAG, "NULL event!!!");
    }
    ESP_LOGE(TAG, "event: %d", event->_event);
    switch (event->_event) {
        case DUER_EVENT_STARTED:
            // Initialize the DCS API
            duer_dcs_init();
            dueros_service_t *serv = audio_service_get_data(duer_serv_handle);
            duer_que_send(serv->duer_que, DUER_CMD_CONNECTED, NULL, 0, 0, 0);
            duer_login_success = true;
            ESP_LOGE(TAG, "event: DUER_EVENT_STARTED");
            xTaskCreate(&report_info_task, "report_info_task", 1024 * 2, NULL, 5, NULL);
            break;
        case DUER_EVENT_STOPPED:
            ESP_LOGE(TAG, "event: DUER_EVENT_STOPPED");
            duer_login_success = false;
            audio_service_disconnect(duer_serv_handle);
            break;
    }
}
```

当event为NULL时为什么不退出，还要继续向下执行？

```
    if (!event) {
        ESP_LOGE(TAG, "NULL event!!!");
        return;
    }
```

jason-mao, 2019-05-24:
@ichenxing Thanks for your report.

ichenxing, 2019-05-30:
@jason-mao duer_event_hook这个函数的问题并没有解决，为什么要关闭呢

jason-mao, 2019-05-30:
@ichenxing 抱歉 漏掉了duer_event_hook的event问题，下次提交修复。

jason-mao, 2019-06-14:
@ichenxing 29678104743d9ce1a86791b9f486d694995a482f 修复了该问题。

closed 2019-06-14 by jason-mao