espressif/esp-adf #174: LyarT4.3板子的codec芯片如何是能L/ROUT2

yelvlab, 2019-05-23:
我在学习adf的时候发现，4.3版本的lyart的板子喇叭声音信号实际上是来自phone_jack的DACOUT信号，而不是LOUT2，然后我找到了`es8388.c`文件，在里面找到了如下代码
```
# line 304
res |= es_write_reg(ES8388_ADDR, ES8388_DACPOWER, tmp);  //0x3c Enable DAC and Enable Lout/Rout/1/2
```

我将其修改为
```
res |= es_write_reg(ES8388_ADDR, ES8388_DACPOWER, 0x3c );  //0x3c Enable DAC and Enable Lout/Rout/1/2
```

然后重新编译了示例代码， 发现信号依旧只有在lout1 rout1输出，lout2 rout2只有一个直流偏置，而没有声音信号，coding能力实在一般，望能多多知道，我是忽略了哪些细节么。

hulkw

jason-mao, 2019-05-24:
@yelvlab 你观察的很细致。DAC可以观察的幅值输出，有多个地方会影响它。一是DAC power开关，还有DAC volume配置。LyraT4.3的驱动默认将L/Rout2的 volume 设置为0，参见`es8388_set_voice_volume`函数。

yelvlab, 2019-05-24:
> @yelvlab 你观察的很细致。DAC可以观察的幅值输出，有多个地方会影响它。一是DAC power开关，还有DAC volume配置。LyraT4.3的驱动默认将L/Rout2的 volume 设置为0，参见`es8388_set_voice_volume`函数。

好，我确认下音volume设定问题，十分感谢

closed 2019-05-24 by jason-mao