espressif/esp-adf #175: Raw stream can't play

xiruilin, 2019-05-26:
Hi,
In <ESP_AUDIO v1.2>, raw stream can not play.
In <media_ctrl_task>:, after four pair <audio_element_run>/<audio_element_resume>,
has "strncasecmp(el->tag, "raw", 3)" function. But now "el->tag" has prefix "IN_" or "OUT_",
so "raw"->"IN_raw", the "strncasecmp()" function is always return non-zero.

jason-mao, 2019-05-27:
@xiruilin Thanks for your report.

jason-mao, 2019-05-27:
@xiruilin Could help to provide your play URI?

xiruilin, 2019-05-27:
Raw stream is used in the "Speak" instruction callback of iFLYOS, 
and the URI is "raw://raw/any.mp3".
(This URI is used for simulation)

easyvolts, 2019-05-28:
Hi, it looks like raw_stream_init function doesn't init ringbuffer of the raw_stream. I'm doing following:

jason-mao, 2019-05-30:
@xiruilin I have test URI works well at 7d6f6ce69318335657733b0bd1566e9bdacd9bf3 as follow.
 - "iis://16000:2@from.pcm/rec.wav#file"
 - "iis://16000:1@record.pcm/record.wav#raw"
 - "raw://record.pcm/record.wav#file"

If you want  to write the raw data to file, should be add fragment filed `#file` at end of URI.

jason-mao, 2019-06-06:
@xiruilin Did you test this issue  as my description?

xiruilin, 2019-06-10:
Sorry, i‘m late.
The raw stream is use to play, not record.
Now，that's how we solved it:

char *audio_element_get_tag(audio_element_handle_t el)
{
    if (counts_resume == 4) {
        counts_resume = 0;
        if (!strcasecmp(el->tag, "IN_raw")) {
            return "raw";
        }
    }
    return el->tag;
}

xiruilin, 2019-06-12:
Https music can't play too.

Error message:

[0;32mI (57412) ESP_AUDIO_TASK: Func:media_ctrl_task, Line:475, MEM Total:3977904 Bytes, Inter:85380 Bytes, Dram:70484 Bytes

[0m
	field_set: 0xb, port: 0
	data[0]: off: 0, len: 5, part: https
	data[1]: off: 8, len: 13, part: cdn.iflyos.cn
	data[2]: unset
	data[3]: off: 21, len: 44, part: /public/open_skills/audio/new-alarm-long.mp3
	data[4]: unset
	data[5]: unset
	data[6]: unset
[0;32mI (57446) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffe20fc[0m
[0;32mI (57454) ESP_AUDIO_CTRL: Func:_ctrl_play, Line:461, MEM Total:3977904 Bytes, Inter:85380 Bytes, Dram:70484 Bytes

[0m
[0;32mI (57454) ESP_AUDIO_TASK: It's a decoder[0m
[0;32mI (57470) ESP_AUDIO_TASK: 1.CUR IN:[IN_https],CODEC:[DEC_mp3],RESAMPLE:[48000],OUT:[OUT_iis][0m
[0;33mW (57479) ESP_AUDIO_TASK: No more in stream handle[0m
[0;32mI (57484) DUEROS_CPP: ESP_AUDIO_CALLBACK_FUNC, st: 5, err: 81001, src: 202[0m

xiruilin, 2019-06-20:
At new esp-adf-libs (ID:ee473006), https music cannot play has fixed.

closed 2019-06-24 by igrr