espressif/esp-adf #177: LyraT v4.3 very low mic gain

LocoMatt, 2019-05-29:
I've got  problem with my LyraT v4.3 board
Recording volume is very low. Only direct and close use of microphone gives barely usable results but still not satisfying. Tried using
`es8388_set_mic_gain(MIC_GAIN_18DB); `
with various values but without succeed.

jason-mao, 2019-05-31:
@LocoMatt The ES8388 have not enough gain for mic, we are set the max gain at default ES8388 driver. If you want to amplify the voice, the ALC can be used, please refer to examples/audio_processing/pipeline_alc

closed 2019-12-11 by jason-mao

expresspotato, 2021-04-10:
For anyone else experiencing this issue, not only is the gain already set to max, but the microphone isn't very good.

You can increase the stream 'volume' with:

`    i2s_alc_volume_set(i2s_stream_reader, 16);`

The scale is from -64 (muted) to 64 (super noisy)

expresspotato, 2021-04-10:
So it appears the ES8388 is doing some sort of 'auto gain' or 'noise floor adjustment' and I'm not sure how to turn it off. You can hear background noise very well when there is no real audio source present in front of the microphone. When that source is introduced, the background noise subsides but do does the audio source and the same time.

Does anyone know how to adjust it?

LocoMatt, 2021-04-10:
> So it appears the ES8388 is doing some sort of 'auto gain' or 'noise floor adjustment' and I'm not sure how to turn it off. You can hear background noise very well when there is no real audio source [...]

I'm unsure, but check page 21 and 22 of ES8388 datasheet.  Register 21, there is "limiter mode". Check also register 19, 20 and 22.

expresspotato, 2021-04-10:
This was actually my mistake, the gain was set so high that is was causing this behavior. Working well now thanks with a bit less gain and i2c_alc_volume.