espressif/esp-adf #180: 能开放设置连接duerOS时候profile信息的能力吗？

MinjunHuang, 2019-06-05:
目前duerOS的连接是将device profile信息拷贝到duer_profile文件中，在编译时候写进去。但是实际的应用，我需要在设备启动时候去获得这个profile信息，然后传给dueros_service执行登录。请问，在不修改dueros_service情况下，怎么实现这个？

jason-mao, 2019-06-06:
@MinjunHuang 可以将profile置于SDCARD/Flash中，然后取读取。Flash的推荐使用partition read的方式

MinjunHuang, 2019-06-07:
@jason-mao 

是可以将profile写入到flash中，但是好像没有看到能够将profile传给dueros_service 中，而dueros_service在duer_login是直接从duer_profile文件中将信息读出来用于登录百度云的。目前我只能修改dueros_serice.c文件，而这个文件是esp-adf的组件，最好是能够不修改它，而能够从外面获取到profile的信息。

ichenxing, 2019-06-09:
```
extern const uint8_t _duer_profile_start[] asm("_binary_duer_profile_start");
extern const uint8_t _duer_profile_end[]   asm("_binary_duer_profile_end");

static void duer_login(void)
{
    int sz = _duer_profile_end - _duer_profile_start;
    char *data = audio_calloc_inner(1, sz);
    if (NULL == data) {
        ESP_LOGE(TAG, "audio_malloc failed");
        return;
    }
    memcpy(data, _duer_profile_start, sz);
    ESP_LOGI(TAG, "duer_start, len:%d\n%s", sz, data);
    duer_start(data, sz);
    audio_free((void *)data);
}
```
修改这一段段代码应该能达到你的要求@MinjunHuang @jason-mao

MinjunHuang, 2019-06-10:
@ichenxing 

1. 是的，我就是在这里修改来实现的，但如果后续这个dueros_service 文件官方修改了的话，我还得同步修改。

2. 但从dueros_service看，是有传用户数据进去的设计的（dueros_service_create函数中有一个user_data参数），只是当前没有接受数据的处理。因此，我才有疑问，是不是有准备将设置profile信息的能力开放出去，这个是有意义的，如果没有，就只能按照你的建议修改。

谢谢你的答复。

closed 2019-06-19 by MinjunHuang