espressif/esp-adf #184: Call esp_audio_play uri(iis://8000:1@from.pcm/rec.amr) , then call esp_audio_stop(player, TERMINATION_TYPE_DONE), then call esp_audio_play error

DanielMengLi, 2019-06-10:
# call esp_audio_play uri(iis://8000:1@from.pcm/rec.amr)  OK
`I (22373) ESP_AUDIO_CTRL: resample_rate:0, channels:1,rate:8000
        field_set: 0x6b, port: 0
        data[0]: off: 0, len: 3, part: iis
        data[1]: off: 13, len: 8, part: from.pcm
        data[2]: unset
        data[3]: off: 21, len: 17, part: /user/rec_000.amr
        data[4]: unset
        data[5]: off: 39, len: 4, part: file
        data[6]: off: 6, len: 6, part: 8000:1
I (22373) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffe0c3c
I (22378) ESP_AUDIO_TASK: It's an encoder
I (22378) ESP_AUDIO_TASK: 1.CUR IN:[IN_iis],CODEC:[ENC_amr],RESAMPLE:[0],OUT:[OUT_file]
I (22378) ESP_AUDIO_TASK: 2.Handles,IN:0x3f807fe0,CODEC:0x3f808cec,FILTER:0x0,OUT:0x3f808708
I (22378) ESP_AUDIO_TASK: 2.2 Update all pipeline
I (22378) ESP_AUDIO_TASK: 2.2 Linked new pipeline
I (22398) AUDIO_PIPELINE: link el->rb, el:0x3f807fe0, tag:IN_iis, rb:0x3f808f70
I (22398) AUDIO_PIPELINE: link el->rb, el:0x3f808cec, tag:ENC_amr, rb:0x3f828fb0
I (22398) ESP_AUDIO_TASK: 3. Previous starting...
I (22398) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:1, Available:5, que:0x3ffe0c3c
I (22403) ESP_AUDIO_TASK: Send MEDIA_CTRL_EVT_PLAY on encoder setup
I (22403) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PLAY, status:0
I (22403) AUDIO_ELEMENT: [IN_iis] Element task created
I (22403) AUDIO_ELEMENT: [OUT_file] Element task created
I (22403) AUDIO_ELEMENT: [ENC_amr] Element task created
I (22408) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:324, MEM Total:4109024 Bytes, Inter:98868 Bytes, Dram:92056 Bytes

I (22408) AUDIO_ELEMENT: [IN_iis] AEL_MSG_CMD_RESUME,state:1
I (22408) I2S_STREAM: AUDIO_STREAM_READER,Rate:8000,ch:1
I (22408) I2S: PLL_D2: Req RATE: 8000, real rate: 8012.000, BITS: 16, CLKM: 78, BCK: 8, MCLK: 2048000.000, SCLK: 256384.000000, diva: 64, divb: 8
I (22413) AUDIO_PIPELINE: Pipeline started
I (22413) AUDIO_ELEMENT: [ENC_amr] AEL_MSG_CMD_RESUME,state:1
I (22413) AMRNB_ENCODER: amrnb open
I (22413) AUDIO_ELEMENT: [OUT_file] AEL_MSG_CMD_RESUME,state:1
I (22418) FILE_STREAM: _file_open,   /sdcard/user/rec_000.amr
I (22418) FILE_STREAM: rec path: /user/rec_000.amr
I (22613) ESP_AUDIO_CTRL: Func:_ctrl_play, Line:443, MEM Total:4102528 Bytes, Inter:98412 Bytes, Dram:91600 Bytes`

# then call esp_audio_stop(player, TERMINATION_TYPE_DONE) OK
`I (24563) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:2, Available:5, que:0x3ffe0c3c
I (24568) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_STOP,data:0x1,size:4
I (24568) AUDIO_ELEMENT: IN-[ENC_amr] AEL_IO_DONE,-2
W (24568) AMRNB_ENCODER: Encoding finish
I (24568) AMRNB_ENCODER: amrnb close
I (24613) AUDIO_ELEMENT: IN-[OUT_file] AEL_IO_DONE,-2`

# then call esp_audio_play ERR
`W (26448) ESP_AUDIO_CTRL: Pipeline is already running. Please stop it first.`

jason-mao, 2019-06-10:
@DanielMengLi Thank you for your report

jason-mao, 2019-11-27:
@DanielMengLi This issue has been fixed on eep-audio ver1.4. Could you help to test?

closed 2019-12-11 by jason-mao