espressif/esp-adf #185: fatal error: Periph_adc_button.h: No such file or directory

ineedsomehelp, 2019-06-11:
When I compile ESP-ADF example ‘play_mp3’，i met this issue。
fatal error: Periph_adc_button.h: No such file or directory
compilation terminated.
~/make/component_wrapper.mk:285: recipe for target 'lyratd_msc_v2_2/board.o' failed
make[1]: *** [lyratd_msc_v2_2/board.o] Error 1
~/make/project.mk:530: recipe for target 'component-audio_board-build' failed

Yuvaraj4041, 2019-06-12:
You need to fetch all submodules by run this command **git submodule update --init**

jason-mao, 2019-06-13:
@ineedsomehelp There is a bug, should be "periph_adc_button.h" at components/audio_board/lyratd_msc_v2_2/board.c and components/audio_board/lyratd_msc_v2_1/board.c.

jason-mao, 2019-06-27:
Has been fixed on ff29502c4b8c6c9dbdc95bb57acf3715aab6a8bb

closed 2019-06-27 by jason-mao