espressif/esp-adf #187: libintl.h：No such file or directory

Gurumin, 2019-06-14:
When I got to the step "Connect and Configure" at https://docs.espressif.com/projects/esp-adf/en/latest/get-started/index.html, it generated a fatal error.
I was using MSYS2 MinGW 32-bit, Thank you for help!
Here is the log:
/usr/bin/gcc -c  -DCURSES_LOC="<curses.h>" -DLOCALE -MMD  /d/Code/esp/esp-adf/esp-idf/tools/kconfig/zconf.tab.c -o zconf.tab.o
In file included from /d/Code/esp/esp-adf/esp-idf/tools/kconfig/zconf.tab.c:91:0:
/d/Code/esp/esp-adf/esp-idf/tools/kconfig/lkc.h:12:11: 致命错误：libintl.h：No such file or directory
 # include <libintl.h>
           ^~~~~~~~~~~
编译中断。
make[1]: *** [Makefile:171：zconf.tab.o] 错误 1
make[1]: 离开目录“/d/Code/esp/esp-adf/esp-idf/tools/kconfig”
make: *** 没有规则可制作目标“/d/Code/esp/esp-adf/esp-idf/tools/kconfig/conf-idf”，由“/d/Code/esp/esp-adf/esp-idf/tools/kconfig/mconf-idf” 需求。 停止。

jason-mao, 2019-06-17:
@Gurumin Please help to confirm  /d/Code/esp/esp-adf/esp-idf/tools/kconfig/lkc.h is it exist?

Gurumin, 2019-06-17:
@jason-mao It exists.

jason-mao, 2019-08-17:
@Gurumin Which version  of IDF and toolchain you used?

jason-mao, 2019-12-11:
@Gurumin How about this issue?

alsaleem00, 2020-08-30:
I got same issue with IDF v4.0. Has this issue been resolved?

jason-mao, 2020-10-15:
@alsaleem00 Could you provide your IDF version and commit id, ADF version and commit id, toolchain version and your error log?

alsaleem00, 2020-10-15:
I have IDF v4.0.1. (4c81978a3e2220674a432a588292a4c860eef27b)
I think the issue is related to MSYS2. I am not using ADF.

closed 2022-02-08 by jason-mao