espressif/esp-adf #188: esp_dispatcher_dueros例程中的audio_setup.c

ichenxing, 2019-06-16:
```
void *setup_player(void *cb, void *ctx)
{
    esp_audio_cfg_t cfg = DEFAULT_ESP_AUDIO_CONFIG();
    audio_board_handle_t board_handle = audio_board_init();
    cfg.vol_handle = board_handle->audio_hal;
    cfg.vol_set = (audio_volume_set)audio_hal_set_volume;
    cfg.vol_get = (audio_volume_get)audio_hal_get_volume;
    cfg.resample_rate = 48000;
    cfg.prefer_type = ESP_AUDIO_PREFER_MEM;
    cfg.cb_func = cb;
    cfg.cb_ctx = ctx;
    esp_audio_handle_t handle = esp_audio_create(&cfg);
    audio_hal_ctrl_codec(board_handle->audio_hal, AUDIO_HAL_CODEC_MODE_BOTH, AUDIO_HAL_CTRL_START);

    // Create readers and add to esp_audio
    fatfs_stream_cfg_t fs_reader = FATFS_STREAM_CFG_DEFAULT();
    fs_reader.type = AUDIO_STREAM_READER;

    esp_audio_input_stream_add(handle, fatfs_stream_init(&fs_reader));
    http_stream_cfg_t http_cfg = HTTP_STREAM_CFG_DEFAULT();
    http_cfg.event_handle = _http_stream_event_handle;
    http_cfg.type = AUDIO_STREAM_READER;
    http_cfg.enable_playlist_parser = true;
    audio_element_handle_t http_stream_reader = http_stream_init(&http_cfg);
    esp_audio_input_stream_add(handle, http_stream_reader);
    // 重复了
    //http_stream_reader = http_stream_init(&http_cfg);
    //esp_audio_input_stream_add(handle, http_stream_reader);
```

jason-mao, 2019-06-17:
@ichenxing 这个不是重复了，这个为了提供两路HTTP下载的时候用的。Pipeline break 一路HTTP的时候，会占用这路HTTP，这时如果再启用播放另外一路HTTP，就会用到。

ichenxing, 2019-06-17:
原来如此，那是我理解错了

closed 2019-06-19 by jason-mao