espressif/esp-adf #189: sonic_set_pitch_and_speed_info(sonic_el, 1.0, 1.0);

NetSensors, 2019-06-16:
The pitch and speed can be increased whilst pipeline is running but you are unable to decrease the pitch and speed whilst pipeline is running.

sonic_set_pitch_and_speed_info(sonic_el, 1.0, 2.0);  works

sonic_set_pitch_and_speed_info(sonic_el, 1.0, 0.5);  does not work

NetSensors, 2019-06-16:
The code change below to **audio_sonic.c** is a quick hack to fix it but takes away any checking of speed and pitch variables.

`esp_err_t sonic_set_pitch_and_speed_info(audio_element_handle_t self, float pitch, float speed)
{
    
    sonic_t *sonic = (sonic_t *)audio_element_getdata(self);
    sonic->sonic_info.speed = speed;
    sonic->sonic_info.speed = pitch;

    
    // sonic_t *sonic = (sonic_t *)audio_element_getdata(self);
    // if ((pitch - 0 < 0.0001)
    //     && (speed - 0 < 0.0001)
    //     && (sonic->sonic_info.pitch - pitch < 0.0001)
    //     && (sonic->sonic_info.speed - speed < 0.0001)) {
    //     return ESP_OK;
    // }
    // if (sonic->sonic_info.pitch - pitch < 0.0001){
    //     if (pitch >= 0.1
    //         && pitch <= 10){
    //         sonic->sonic_info.pitch = pitch;
    //         ESP_LOGI(TAG, "reset pitch of stream data : %.2f", sonic->sonic_info.pitch);
    //     }else{
    //         ESP_LOGD(TAG, "The pitch must be in [0.1 10],reset pitch of stream data : %.2f", pitch);
    //         return ESP_FAIL;
    //     } 
    // }
    // if (sonic->sonic_info.speed - speed < 0.0001){
    //     if (speed >= 0.1
    //         && speed <= 10){
    //         sonic->sonic_info.speed = speed;   
    //         ESP_LOGI(TAG, "reset speed of stream data : %.2f", sonic->sonic_info.speed);
    //     }else{
    //         ESP_LOGD(TAG, "The speed must be in [0.1 10],reset speed of stream data : %.2f", speed);
    //         return ESP_FAIL;
    //     } 
    // }  
    return ESP_OK;
}
`

jason-mao, 2019-06-18:
@NetSensors Thanks for your report.

closed 2019-06-24 by igrr