espressif/esp-adf #191: play_mp3_control example doesn't work

vgonet, 2019-06-18:
Hello,

I'm trying to run the example  esp-adf/examples/get-started/play_mp3_control on my ESP32 LyraT **V4.3**.

The ESP-ADF version I use is **V1.0**

I've the following issue (gpio_install_isr_service(394): GPIO isr service already installed):

I (319) cpu_start: Pro cpu start user code
I (112) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (113) PLAY_MP3_FLASH: [ 1 ] Start audio codec chip
I (138) PLAY_MP3_FLASH: [ 2 ] Create audio pipeline, add all elements to pipeline, and subscribe pipeline event
I (138) PLAY_MP3_FLASH: [2.1] Create mp3 decoder to decode mp3 file and set custom read callback
I (147) PLAY_MP3_FLASH: [2.2] Create i2s stream to write data to codec chip
I (157) PLAY_MP3_FLASH: [2.3] Register all elements to audio pipeline
I (162) PLAY_MP3_FLASH: [2.4] Link it together [mp3_music_read_cb]-->mp3_decoder-->i2s_stream-->[codec_chip]
I (172) PLAY_MP3_FLASH: [ 3 ] Initialize peripherals
**E (178) gpio: gpio_install_isr_service(394): GPIO isr service already installed**
I (186) PLAY_MP3_FLASH: [3.1] Initialize Touch peripheral
I (192) PLAY_MP3_FLASH: [3.2] Start all peripherals
W (214) PERIPH_TOUCH: _touch_init
I (214) PLAY_MP3_FLASH: [ 4 ] Setup event listener
I (214) PLAY_MP3_FLASH: [4.1] Listening event from all elements of pipeline
I (220) PLAY_MP3_FLASH: [4.2] Listening event from peripherals
W (226) PLAY_MP3_FLASH: [ 5 ] Tap touch buttons to control music player:
W (234) PLAY_MP3_FLASH:       [Play] to start, pause and resume, [Set] to stop.
W (241) PLAY_MP3_FLASH:       [Vol-] or [Vol+] to adjust volume.

I don't know who initialize the GPIO isr service...

Best regards,
Vincent

withdra, 2019-06-18:
Hi vgonet, this line of logs is normal, because in audio_hal_ctrl_codec() and esp_periph_set_init() both install the gpio ISR. It does not affect usage. You should connect a speaker and then touch the "play" pad to play music.

withdra, 2019-06-18:
@vgonet

closed 2019-07-29 by jason-mao