espressif/esp-adf #193: 采用duerOS例程播放新闻大约3分钟左右就系统重启

yqj2012, 2019-06-19:
问题描述，
   采用dueros例程，添加自己的Profile文件后开始直接编译烧录后系统卡在sd卡的位置， 由于我暂时没有sd卡， 我就直接跳过那一个循环的位置，系统正常跑起来了，由于我配置了默认wifi网络，系统能自动连接上dueros。
  对板子说：“播放新闻”，板子开始播放新闻，大约3分钟左右重启，日志如下：
I (160106) AUDIO_WRAPPER: Get play position 0 of 0
I (160120) AUDIO_WRAPPER: Get play position 0 of 0
E (160122) AUDIO_WRAPPER: duer_dcs_audio_on_finished, 2
I (160443,tid:3ffdf31c) lightduer_dcs_router.c( 407):Directive name: Play
I (160443,tid:3ffdf31c) lightduer_dcs_audio_player.c( 571):token: eyJib3RfaWQiOiJhdWRpb19uZXdzIiwicmVzdWx0X3Rva2VuIjoiZjY2NmMzMTFmYWY0ZmM3ZDY2NzljZTY3NjMwZDY2NjAiLCJib3RfdG9rZW4iOiIxMDE4MTU4Njk4ODkyODY5Nzk0NCIsImxhdW5jaF9pZHMiOlsiYXVkaW9fbmV3cyJdfQ==
I (160542) AUDIO_WRAPPER: Playing audio, offset:0 url:http://xiaodu.baidu.com/speech/api/media?resource_id=785cd75f160b8852b3099b6a1be9cd3b&token=4e110745095a6a50011a0e461c57554b1a4640430300411c46434756014d5710721206015e1d1b1d170941124811564c0c5c165e0f5507431c444443050353005403541b491a570844010d07721d10510f1132490158027e2470643575230c011f0735057d0254461d026652365f03091b0b415f0f12061c581159120a56424a4745&key=v2&stream=false
W (160580) ESP_AUDIO_CTRL: This URI has no extension and decoder type is not 'auto'
        field_set: 0x1b, port: 0
        data[0]: off: 0, len: 4, part: http
        data[1]: off: 7, len: 16, part: xiaodu.baidu.com
        data[2]: unset
        data[3]: off: 23, len: 17, part: /speech/api/media
        data[4]: off: 41, len: 331, part: resource_id=785cd75f160b8852b3099b6a1be9cd3b&token=4e110745095a6a50011a0e461c57554b1a4640430300411c46434756014d5710721206015e1d1b1d170941124811564c0c5c165e0f5507431c444443050353005403541b491a570844010d07721d10510f1132490158027e2470643575230c011f0735057d0254461d026652365f03091b0b415f0f12061c581159120a56424a4745&key=v2&stream=false
        data[5]: unset
        data[6]: unset
I (160640) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffe0740
I (160647) ESP_AUDIO_TASK: It's a decoder
I (160651) ESP_AUDIO_TASK: 1.CUR IN:[IN_http],CODEC:[NULL],RESAMPLE:[48000],OUT:[OUT_iis]
I (160660) AUDIO_ELEMENT: [IN_http] RESUME: Element has finished, state:5
I (160668) ESP_AUDIO_CTRL: Func:_ctrl_play, Line:461, MEM Total:4152896 Bytes, Inter:76428 Bytes, Dram:42452 Bytes

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x401f1363  PS      : 0x00060930  A0      : 0x8012b3d8  A1      : 0x3ffc3cd0  
A2      : 0x00000000  A3      : 0x3ffc3d10  A4      : 0x3ffe462c  A5      : 0x00000000  
A6      : 0x00000bb8  A7      : 0x00000001  A8      : 0x8008446f  A9      : 0x3ffc3cb0  
A10     : 0x00027f54  A11     : 0x3ffc3cc0  A12     : 0x3ffb12f8  A13     : 0x3ffc3cb0  
A14     : 0x000000fe  A15     : 0x3ffb4730  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000054  LBEG    : 0x40096dd0  LEND    : 0x40096ddb  LCOUNT  : 0x00000000  

Backtrace: 0x401f1363:0x3ffc3cd0 0x4012b3d5:0x3ffc3cf0

yqj2012, 2019-06-19:
说“播放相声”，系统不重启，但是播放时间长后就不能播放了，百度后台能收到命令，但是没有声音输出

xiruilin, 2019-06-19:
I got the same issue!
I found that it happened in the continuous playback of HTTP streams without extensions.
The message log:

W (7335609) ESP_AUDIO_CTRL: This URI has no extension and decoder type is not 'auto'
        field_set: 0x1b, port: 0
        data[0]: off: 0, len: 4, part: http
        data[1]: off: 7, len: 16, part: xiaodu.baidu.com
        data[2]: unset
        data[3]: off: 23, len: 17, part: /speech/api/media
        data[4]: off: 41, len: 327, part: resource_id=2353f37384504eac35de168841bc952d&token=4e110745095a6a50011a0e461451504b154542450c114f1214415a5d105b403b5b011100441b1a510f11411c465
04050011a0e464f5006451f10154700065301461f175a0951510a593b0a170f4e562a6c602476280451000f56041e690b101449333f2b7e452a0834020253125d6646014605014219564917520f51165e1744&key=v2&stream=fals
e
        data[5]: unset
        data[6]: unset
I (7335661) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffe26c4
I (7335669) ESP_AUDIO_CTRL: Func:_ctrl_play, Line:461, MEM Total:3967440 Bytes, Inter:77764 Bytes, Dram:62512 Bytes

I (7335669) ESP_AUDIO_TASK: It's a decoder
I (7335685) ESP_AUDIO_TASK: 1.CUR IN:[IN_http],CODEC:[NULL],RESAMPLE:[48000],OUT:[OUT_iis]
I (7335695) AUDIO_ELEMENT: [IN_http] Element task created
I (7335700) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:1
[events-response][sid:131] 204
[events-response][sid:131] stream close
I (7336092) HTTP_STREAM: total_bytes=9576
I (7336092) AUDIO_ELEMENT: OUT-[IN_http] AEL_IO_DONE,0
I (7336095) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_PAUSE
I (7336097) ESP_AUDIO_TASK: 2.Handles,IN:0x3f80c044,CODEC:0x3f80c8a0,FILTER:0x3f815b64,OUT:0x3f80cc64
W (7336108) AUDIO_PIPELINE: audio_pipeline state is not running, 1
W (7336114) AUDIO_PIPELINE: audio_pipeline state is not running, 1
I (7336121) AUDIO_PIPELINE: audio_pipeline_unlinked
W (7336126) AUDIO_PIPELINE: There are no listener registered
W (7336132) ESP_AUDIO_TASK: Destroy the old pipeline
I (7336138) ESP_AUDIO_TASK: 2.2 Update all pipeline
I (7336144) ESP_AUDIO_TASK: 2.3 Linked new pipeline
I (7336150) AUDIO_PIPELINE: link el->rb, el:0x3f80c044, tag:IN_http, rb:0x3f815b38
I (7336158) AUDIO_PIPELINE: link el->rb, el:0x3f80c8a0, tag:DEC_mp3, rb:0x3f815a58
I (7336166) AUDIO_PIPELINE: link el->rb, el:0x3f815b64, tag:resample, rb:0x3f815a94
I (7336174) ESP_AUDIO_TASK: 3. Previous starting...
I (7336180) AUDIO_ELEMENT: [DEC_mp3] Element task created
I (7336185) AUDIO_ELEMENT: [DEC_mp3] AEL_MSG_CMD_RESUME,state:1
I (7336186) AUDIO_ELEMENT: [resample] Element task created
I (7336192) STAGEFRIGHTMP3_DECODER: MP3 opened
I (7336198) AUDIO_ELEMENT: [OUT_iis] Element task created
I (7336209) AUDIO_ELEMENT: [OUT_iis] AEL_MSG_CMD_RESUME,state:1
I (7336216) I2S_STREAM: AUDIO_STREAM_WRITER
I (7336199) AUDIO_ELEMENT: [resample] AEL_MSG_CMD_RESUME,state:1
I (7336228) RSP_FILTER: sample rate of source data : 16000, channel of source data : 1, sample rate of destination data : 48000, channel of destination data : 2
I (7336199) AUDIO_ELEMENT: [IN_http] AEL_MSG_CMD_RESUME,state:5
I (7336309) HTTP_STREAM: total_bytes=9576
I (7336319) DUEROS_CPP: ESP_AUDIO_CALLBACK_FUNC, st: 1, err: 0, src: 202
I (7336324) STAGEFRIGHTMP3_DECODER: I2S setup: sample_rate = 16000 : channels = 1
I (7336325) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:1, Available:5, que:0x3ffe26c4
I (7336332) ESP_AUDIO_TASK: Send MEDIA_CTRL_EVT_PLAY on elements report info
I (7336340) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PLAY, status:1
W (7336346) AUDIO_PIPELINE: Pipeline already started, state:2
W (7336372) HTTP_STREAM: No more data,errno:0, total_bytes:9576
W (7336372) HTTP_STREAM: there are no track
I (7336373) AUDIO_ELEMENT: IN-[IN_http] AEL_IO_DONE,0
I (7340038) AUDIO_ELEMENT: IN-[DEC_mp3] AEL_IO_DONE,-2
I (7341119) STAGEFRIGHTMP3_DECODER: MP3 decodig done
I (7341120) STAGEFRIGHTMP3_DECODER: Closed
I (7341276) AUDIO_ELEMENT: IN-[resample] AEL_IO_DONE,-2
I (7341277) RSP_FILTER: rsp_filter_close
I (7341302) AUDIO_ELEMENT: IN-[OUT_iis] AEL_IO_DONE,-2
I (7341342) DUEROS_CPP: ESP_AUDIO_CALLBACK_FUNC, st: 4, err: 0, src: 202
W (7341342) ESP_AUDIO_CTRL: This URI has no extension and decoder type is not 'auto'
        field_set: 0x1b, port: 0
        data[0]: off: 0, len: 4, part: http
        data[1]: off: 7, len: 16, part: xiaodu.baidu.com
[... 13 lines cut ...]
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40170ff4  PS      : 0x00060830  A0      : 0x80139519  A1      : 0x3ffc5c80
0x40170ff4: audio_element_get_tag at E:/esp-adf/components/audio_pipeline/audio_element.c:763

A2      : 0x00000000  A3      : 0x3ffc5cc0  A4      : 0x3ffe3170  A5      : 0x00000000
A6      : 0x00000bb8  A7      : 0x00000001  A8      : 0x3ffc4648  A9      : 0x00000008
A10     : 0x00701135  A11     : 0x3ffc5c6c  A12     : 0x3ffbea94  A13     : 0x3ffc5c78
A14     : 0x000000fe  A15     : 0x3ffc3ca0  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000054  LBEG    : 0x4009521c  LEND    : 0x40095227  LCOUNT  : 0x00000000
0x4009521c: memset at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/mem
set.S:142

0x40095227: memset at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/mem
set.S:152

ELF file SHA256: 4104f4e076e74819896f10feeb3dce7364915201424cd80f12c6dde31002ee8e

Backtrace: 0x40170ff4:0x3ffc5c80 0x40139516:0x3ffc5ca0
0x40170ff4: audio_element_get_tag at E:/esp-adf/components/audio_pipeline/audio_element.c:763

0x40139516: get_codec_type_by_instream at /Users/maojianxin/duros/esp-adf-libs-source/esp_audio/src/media_ctrl_task.c:173
 (inlined by) media_ctrl_task at /Users/maojianxin/duros/esp-adf-libs-source/esp_audio/src/media_ctrl_task.c:341

Rebooting...

jason-mao, 2019-06-20:
@yqj2012 Thanks for your report.

yqj2012, 2019-06-20:
I have two ESP32-LyraT boards, one of which has no such problem.

xiruilin, 2019-06-20:
At new esp-adf-libs (ID:1c0aa42c), play HTTP streams without extensions is still error. First play good,
and next it timeout. The message log:

W (41648) ESP_AUDIO_CTRL: This URI has no extension and decoder type is not 'auto'
I (41649) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffdc728
I (41656) ESP_AUDIO_TASK: It's a decoder
I (41661) ESP_AUDIO_TASK: 1.CUR IN:[IN_http],CODEC:[NULL],RESAMPLE:[48000],OUT:[OUT_iis]
W (41673) ESP_AUDIO_TASK: Try to got muisc format by in stream.
I (42087) ESP_AUDIO_TASK: 2.Handles,IN:0x3f80d674,CODEC:0x3f80ded0,FILTER:0x3f80eb5c,OUT:0x3f80e294
W (42088) ESP_AUDIO_TASK: Destroy the old pipeline
I (42091) ESP_AUDIO_TASK: 2.2 Update all pipeline
I (42096) ESP_AUDIO_TASK: 2.3 Linked new pipeline
I (42104) ESP_AUDIO_TASK: 3. Previous starting...
I (42107) STAGEFRIGHTMP3_DECODER: MP3 opened
I (42194) STAGEFRIGHTMP3_DECODER: I2S setup: sample_rate = 16000 : channels = 1
I (42195) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:1, Available:5, que:0x3ffdc728
I (42199) ESP_AUDIO_TASK: Send MEDIA_CTRL_EVT_PLAY on elements report info
I (42206) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PLAY, status:1
I (46847) STAGEFRIGHTMP3_DECODER: MP3 decodig done
I (46847) STAGEFRIGHTMP3_DECODER: Closed
W (47069) ESP_AUDIO_CTRL: This URI has no extension and decoder type is not 'auto'
I (47069) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffdc728
I (47074) ESP_AUDIO_TASK: It's a decoder
I (47079) ESP_AUDIO_TASK: 1.CUR IN:[IN_http],CODEC:[NULL],RESAMPLE:[48000],OUT:[OUT_iis]
W (47088) ESP_AUDIO_TASK: Try to got muisc format by in stream.
W (55075) ESP_AUDIO_CTRL: Play activity timeout
W (55076) ESP_AUDIO_CTRL: This URI has no extension and decoder type is not 'auto'
I (55077) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffdc728
W (63086) ESP_AUDIO_CTRL: Play activity timeout
W (63086) ESP_AUDIO_CTRL: This URI has no extension and decoder type is not 'auto'
I (63088) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:4, que:0x3ffdc728
W (71096) ESP_AUDIO_CTRL: Play activity timeout
W (71096) ESP_AUDIO_CTRL: This URI has no extension and decoder type is not 'auto'
I (71098) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:3, que:0x3ffdc728
W (79106) ESP_AUDIO_CTRL: Play activity timeout

jason-mao, 2019-06-20:
@yqj2012 Thanks. Got that.

jason-mao, 2019-06-20:
@yqj2012 What did you say to reproduce this issue?

yqj2012, 2019-06-20:
@jason-mao Say to the board: "Play the news"

yqj2012, 2019-06-20:
@jason-mao Say to the board: "Play the crosstalk", this will play for a long time, but the time will not play the sound, but Baidu background data can see the recognition of the voice.

jason-mao, 2019-06-20:
@yqj2012 Thank you. Let me try to reproduced.

yqj2012, 2019-06-20:
Thank you very much, look forward to the solution.

closed 2019-06-26 by igrr

yqj2012, 2019-06-26:
The problem is solved? I am going to try this code node. Thank you!