espressif/esp-adf #194: fatal error: audio_element.h: No such file

lin-zi-jun, 2019-06-24:
I'm finding this file in path esp-adf/components/audio_pipeline/include/audio_element.h.
But there was an error when I compiled it.
Error is as follows:
CC build/main/play_mp3_example.o
/home/jun/esp/play_mp3/main/play_mp3_example.c:15:10: fatal error: audio_element.h: No such file or directory

vikramdattu, 2019-06-24:
Did you forgot to provide ADF_PATH?

lin-zi-jun, 2019-06-25:
I compiled esp_idf  helloword examples OK，
But I compile esp_adf  play_mp3 examples  Prompt error.
The environment path I configured is as follows:

# set PATH so it includes user's private bin directories
PATH="$HOME/bin:$HOME/.local/bin:$PATH"
export PATH="$HOME/esp/xtensa-esp32-elf/bin:$PATH"
export IDF_PATH="~/esp/esp-idf"
export ADF_PATH="~/esp/esp-adf"

whether make menuconfig and need to select the audio board?
But I didn't find the choice.

lin-zi-jun, 2019-06-25:
I think I found the answer. I added double quotes to the path

closed 2019-06-25 by lin-zi-jun

WHUWeiYanLin, 2024-08-02:
I'd like to know how you added it.

WHUWeiYanLin, 2024-08-02:
I'd like to know how you added it, any examples to show?