espressif/esp-adf #195: node ‘’adc6ac1’’ compiles 'dueros' error

yqj2012, 2019-06-27:
Hello, I am compiling the error when updating the code to the latest node ‘’adc6ac1’’. The log is as follows: 
/mnt/g/esp/esp-adf/components/esp-adf-libs/esp_sr/lib/libesp_wakenet.a(wakenet5_quantized.o):(.literal.model_init+0x0): undefined reference to `get_coeff_wakeNet5_model_quant'
collect2: error: ld returned 1 exit status
/mnt/g/esp/esp-adf/esp-idf/make/project.mk:460: recipe for target '/mnt/g/esp/esp-adf/dueros/build/esp_dueros.elf' failed
make: *** [/mnt/g/esp/esp-adf/dueros/build/esp_dueros.elf] Error 1

yqj2012, 2019-06-27:
Just tried because the configuration is "Speech Recognition Configuration --->Wakeup word is Alexa" will be wrong, and configured as "Wakeup word is Hi Lexin" will not

jason-mao, 2019-06-27:
@yqj2012 Alexa only support WakeNet3.

yqj2012, 2019-06-27:
You can do this in the Kconfig file, and avoid the same mistakes that others will download and compile. Thank you.

jason-mao, 2019-06-27:
Thank you mentioned that.

closed 2019-06-27 by jason-mao