espressif/esp-adf #200: A documentation problem

jj11hh, 2019-06-30:
At the Setup ESP-IDF section of the ESP-ADF documentation (https://docs.espressif.com/projects/esp-adf/en/latest/get-started/#setup-esp-idf)
the `ESP32 Documentation' link refers to the latest (v4.0-dev) ESP-IDF documentation,
it tells me to install the latest toolchain (gcc-8.2.0) (https://docs.espressif.com/projects/esp-idf/en/latest/get-started/linux-setup.html)
but, it seems ESP-ADF only supports ESP-IDF 3.2 and toolchain gcc-5.2.0, follow this tutorial, I installed toolchain 8.2.0 and it's totally unusable due to lots of compilation error.

So I think this link should be corrected to the STABLE documentation rather than the LATEST one:
https://docs.espressif.com/projects/esp-adf/en/stable/get-started/#setup-esp-idf

I have really got confused by this issue. here's the error output.

> Toolchain path: /home/ihen/esp32/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: esp32-2019r1
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 8.2.0
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Toolchain version is not supported: esp32-2019r1
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 8.2.0
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
CC build/bootloader/bootloader_support/src/bootloader_flash.o
CC build/bootloader/bootloader_support/src/flash_qio_mode.o
CC build/bootloader/bootloader_support/src/efuse.o
CC build/bootloader/bootloader_support/src/bootloader_init.o
CC build/bootloader/bootloader_support/src/bootloader_common.o
CC build/bootloader/bootloader_support/src/secure_boot.o
CC build/bootloader/bootloader_support/src/bootloader_clock.o
CC build/bootloader/bootloader_support/src/bootloader_random.o
CC build/bootloader/bootloader_support/src/secure_boot_signatures.o
CC build/bootloader/bootloader_support/src/flash_partitions.o
CC build/bootloader/bootloader_support/src/esp_image_format.o
CC build/bootloader/bootloader_support/src/bootloader_sha.o
CC build/bootloader/bootloader_support/src/flash_encrypt.o
CC build/bootloader/bootloader_support/src/bootloader_utility.o
AR build/bootloader/bootloader_support/libbootloader_support.a
CC build/bootloader/log/log.o
AR build/bootloader/log/liblog.a
CC build/bootloader/spi_flash/spi_flash_rom_patch.o
AR build/bootloader/spi_flash/libspi_flash.a
CC build/bootloader/micro-ecc/micro-ecc/uECC.o
AR build/bootloader/micro-ecc/libmicro-ecc.a
CC build/bootloader/soc/esp32/soc_memory_layout.o
CC build/bootloader/soc/esp32/rtc_pm.o
CC build/bootloader/soc/esp32/rtc_periph.o
CC build/bootloader/soc/esp32/cpu_util.o
CC build/bootloader/soc/esp32/sdmmc_periph.o
CC build/bootloader/soc/esp32/rtc_wdt.o
CC build/bootloader/soc/esp32/sdio_slave_periph.o
CC build/bootloader/soc/esp32/rtc_init.o
CC build/bootloader/soc/esp32/spi_periph.o
CC build/bootloader/soc/esp32/rtc_clk_init.o
CC build/bootloader/soc/esp32/gpio_periph.o
CC build/bootloader/soc/esp32/rtc_sleep.o
CC build/bootloader/soc/esp32/rtc_time.o
CC build/bootloader/soc/esp32/rtc_clk.o
CC build/bootloader/soc/src//memory_layout_utils.o
AR build/bootloader/soc/libsoc.a
CC build/bootloader/main/bootloader_start.o
AR build/bootloader/main/libmain.a
LD build/bootloader/bootloader.elf
esptool.py v2.6
Python requirements from /home/ihen/esp32/esp-adf/esp-idf/requirements.txt are satisfied.
Building partitions from /home/ihen/esp32/esp-adf/esp-idf/components/partition_table/partitions_singleapp.csv...
CC build/adf_utils/json_utils.o
CC build/adf_utils/cloud_services/baidu_access_token.o
CC build/adf_utils/cloud_services/aws_sig_v4_signing.o
AR build/adf_utils/libadf_utils.a
CC build/app_trace/app_trace_util.o
CC build/app_trace/host_file_io.o
CC build/app_trace/app_trace.o
CC build/app_trace/gcov/gcov_rtio.o
AR build/app_trace/libapp_trace.a
CC build/app_update/esp_ota_ops.o
AR build/app_update/libapp_update.a
CXX build/asio/asio/asio/src/asio.o
In file included from /home/ihen/esp32/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/cstdlib:75,
                 from /home/ihen/esp32/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/ext/string_conversions.h:41,
                 from /home/ihen/esp32/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/bits/basic_string.h:6391,
                 from /home/ihen/esp32/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/string:52,
                 from /home/ihen/esp32/esp-adf/esp-idf/components/asio/asio/asio/include/asio/impl/error.ipp:19,
                 from /home/ihen/esp32/esp-adf/esp-idf/components/asio/asio/asio/include/asio/impl/src.hpp:22,
                 from /home/ihen/esp32/esp-adf/esp-idf/components/asio/asio/asio/src/asio.cpp:11:
/home/ihen/esp32/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:155:44: error: expected initializer before '__result_use_check'
 void *reallocarray(void *, size_t, size_t) __result_use_check __alloc_size(2)
                                            ^~~~~~~~~~~~~~~~~~
/home/ihen/esp32/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:340:52: error: expected initializer before '__alloc_align'
 void * aligned_alloc(size_t, size_t) __malloc_like __alloc_align(1)
                                                    ^~~~~~~~~~~~~
make[1]: *** [/home/ihen/esp32/esp-adf/esp-idf/make/component_wrapper.mk:286：asio/asio/src/asio.o] 错误 1
make: *** [/home/ihen/esp32/esp-idf/make/project.mk:530：component-asio-build] 错误 2

krzychb, 2019-07-02:
Hi @jj11hh this is good point. Sorry for that. We will fix documentation.

closed 2019-07-09 by igrr

NickVerlooy93, 2019-07-11:
Is this fixed? I'm experiencing the same. 
Can you please point out a bit more in detail how to fix this?