espressif/esp-adf #207: Build of advanced_examples/http_play_and_save_to_file fails

easyvolts, 2019-07-01:
Hi,
when I try building advanced_examples/http_play_and_save_to_file project it fails with following problem:

D:/GitRepo/esp-adf/components/audio_stream/http_stream.c: In function '_http_open':
D:/GitRepo/esp-adf/components/audio_stream/http_stream.c:501:9: error: implicit declaration of function 'esp_http_client_set_redirection' [-Werror=implicit-function-declaration]
         esp_http_client_set_redirection(http->client);
         ^
cc1.exe: some warnings being treated as errors

vikramdattu, 2019-07-01:
Please change idf to release/v3.2

Also unlike previous method, if only you haven't set IDF_PATH, adf will consider it as <adf_path>/esp-idf/

easyvolts, 2019-07-11:
Thank you for the explanation, removing IDF_PATH helped and now example can be built as expected.

closed 2019-07-11 by easyvolts