espressif/esp-adf #208: TLS support in VOIP/SIP

songshan99, 2019-07-02:
Hi, 
firstly thanks for the VOIP example in the new release. It works well with mainstream VOIP servers.

By looking at the code, I believe the existing example is using non-encrypted traffic, not over TLS, and TLS is currently not supported in the ESP SIP library. Is this the correct statement? please help me verify.

And meanwhile I'd like to ask if you have any plan to support TLS in VOIP in the near future. Thanks!

Regards
Shan

tuanpmt, 2019-07-02:
Hi @songshan99 
You can try TLS with the URI = `tls://user:pass@ip:port` and TCP with the URI = `tcp://user:pass@ip:port`

songshan99, 2019-07-02:
Thanks for the explanation!

songshan99, 2019-07-02:
Great. The TLS and TCP works as you explained. Thanks.

Now my next question is: How to add a customized client certificate to the ESP32 SIP application so that the server can verify it?

closed 2019-08-20 by igrr

Patsjemoe, 2019-08-22:
Hi,
Can anyone please confirm how you initiate a call ? When the program is flashed, I push the hard reset button....but what then ? 
regards

Patsjemoe, 2019-08-22:
Hi,
tried TLS, and get following error : 
 (201873) SIP: Conecting...
E (205913) esp-tls: mbedtls_ssl_handshake returned -0x4c
E (205913) esp-tls: Failed to open new connection
E (205913) TRANS_SSL: Failed to open a new connection
E (205913) SIP: Connection failed, error: Bad file number
how to solve ?

songshan99, 2019-08-23:
Your server enabled TLS at the correct port? This was the error I got when the FreePBX is not configured correctly.

Patsjemoe, 2019-08-23:
Hi,
I tried TLS without succes, so I moved back to TCP, and the error I receive
now is with using TCP.....
Is my assumption correct that authentication should only happen when using
TLS, not TCP ?
I am not using PBX, but trying to make contact between an android app from
Linphone with SIP and the ESP32. In the app, I can set TCP, UDP or TLS.

songshan99, 2019-08-23:
Authentication happens no matter it is TLS or TCP or UDP. 

You have to setup a SIP server between your ESP32 and your Android APP. That's what freePBX does.

Patsjemoe, 2019-08-23:
Hi,
Thanks, I am new to VOIP, and was not aware that freepbx was required...
Have installed a freepbx on a raspberry pi.
Unfortunately, it is so unclear on how to setup....do you know a simple
instruction manual that tells you how to
use the sip identity + local address to setup freepbx ? it always gives
errors and changes ports from 5060 to 5061 and 5062...
regards
Ludo

Patsjemoe, 2019-08-29:
Dear Songshan,
can you confirm if this application really works ?
I have my sdk running ok now, but I keep getting problems.
I succeed to log in, but then it makes a reread, and get an error.
SO it looks as the esp32 program in github is not working, but it seems to
be impossible to have anyone
to confirm if that really works, or if it is not complete.
This is important to know, because I had no experience with VOIP and
freepbx and esp32, so to know if
the problem is because I do something stupid in freepbx or if the problem
is because the program is not
complete...Now, I am just trying in the dark.
regards
Ludo

songshan99, 2019-08-29:
I've tested it using latest FreePBX and ESP-ADF Pre-release v2.0-beta1. 
I'm running freePBX from virtual machine.
There isn't a big issue with ESP-ADF code.
There are a couple of difficulties when setting up the network, the FreePBX server, routers and firewalls. I'd suggest you focus on these things since ESP-ADF code just works fine as proven to me.

Patsjemoe, 2019-08-29:
Hi,
thank you, so I know it is not the code.

I was getting too may errors of missing files when I tried version 2.0,
therefore, I had to stay with the latest stable verison, not beta.

Is it tested with v3.2.2 ? (if not, could you do that ?)
Is there any reason why it would not work with v3.2.2 ?

could you confirm where the difficulties with freepbx mainly were, and what
the correct solution was ? It may help to get errors deleted....

regards
Ludo

Patsjemoe, 2019-08-29:
Hi,
Can I ask another question :
where in the program is the read again done ?  (I do not find this in the
code, is this done in a sip driver ?)
This is where I get my error :

W (8003) SIP: Read again
W (12003) SIP: Error read or timeout
E (12003) SIP: Error register device
sorry for the questions, but try to move forward and I am stuck....
regards
Ludo

songshan99, 2019-08-29:
It might be in the binary library so you don't see the source code

Patsjemoe, 2019-08-29:
can you verify and confirm in which file this can be found ?
then I can check what it really does....

songshan99, 2019-08-29:
There is a libesp_sip.a under components/esp_adf_lib.

The true function of VoIP is implemented here.

Patsjemoe, 2019-08-30:
Hi,
thank you, found it.
Is your test using TLS ?
as I get following error, where should I look for as cause of the problem ?
W (8013) SIP: Read again
W (12013) SIP: Error read or timeout
E (12013) SIP: Error register device
W (12013) SIP: CHANGE STATE FROM 1, TO 0, :func: sip_register:1334

songshan99, 2019-08-30:
I tested TLS TCP and UDP. From the log you sent, it seems that the device
failed to register. I suggest use a PC VoIP software such as microSIP to
make sure your freepbx is set up correctly.

Patsjemoe, 2019-08-30:
thank you for your reply.

Patsjemoe, 2019-08-30:
installed microsip.
get a "request timeout"...but microsip site does not provide info on this
error...

songshan99, 2019-08-30:
It looks straight forward that the freepbx server is not reachable.

Patsjemoe, 2019-08-30:
but running esp32, I had first
I (765963) SIP: Conecting...
W (765963) SIP: CHANGE STATE FROM 0, TO 1, :func: sip_connect:1319
W (765963) SIP: [1970-01-01/00:12:22]=======WRITE 0599 bytes>>
I (765973) SIP:

before the read againg
W (8013) SIP: Read again
W (12013) SIP: Error read or timeout
E (12013) SIP: Error register device
W (12013) SIP: CHANGE STATE FROM 1, TO 0, :func: sip_register:1334

...I would think that it means it reached the freepbx...or is this only
linphone provider ?

songshan99, 2019-08-30:
It says connecting, not connected. So there is no sign that the esp32 is
connected to the server successfully

Patsjemoe, 2019-08-31:
Hi,
I still have problems to move forward.
Can Espressif not provide a 2 page document, showing exactly what needs to
be put in freepbx, mainly explaining(showing with picture) if it is
provider info, or own network info, which passwords : provider or
freepbx....etc ?
that would be helpfull.
Which company did you get the SIP username and identity from ? I have used
linphone. ANy experience with that ?
What also would help is to have an SIP account that we can call, so we know
at least that the other site is ok ...
Anything that can help me, also helps espressif ....
regards
Ludo

Patsjemoe, 2019-09-01:
Hi,
I have freepbx running now with microsip. Thank you for the suggestion.

But when running esp32, I still get the same error when using port 5061.
If I use port 5060, I get a SIP: Required authentication error.
(In menuconfig : udp://username:password@sip.linphone.org:5060)

Regards
LUdo

songshan99, 2019-09-01:
The user name and password shall be configured the same as the account
setup in freepbx.

Patsjemoe, 2019-09-01:
Hi,
It is exactly the same as in PEER in the TRUNK.
It is exactly the same as I put in menuconfig.

Patsjemoe, 2019-09-01:
Hi,
when I use wireshark when making a phonecall with microsip, I see at a
moment ....:  status : 407 Proxy Authentication Required , and the proxy is
acknowledged.
As I have also have  this SIP: Required authentication problem, I wonder
how ESP32 VOIP is providing that information, because I cannot put that in
menuconfig.
regards
Ludo

songshan99, 2019-09-01:
Do you use the same username password for both esp32 and microsip? You have
to set up 2 different numbers so they can call each other.

Patsjemoe, 2019-09-01:
I am using both esp32 as microsip to call to my android phone where I
installed linphone. So, calling my linphone with microsip works, and with
the same data in esp32, I cannot get connected with android linphone.

songshan99, 2019-09-01:
What is the information printed out on esp32 debug UART?

Since it's the same number on both microsip and esp32, it's possible esp32
does not log in correctly. Please watch out for the logging in information.

Meanwhile, can you call from linphone to esp32?

Patsjemoe, 2019-09-01:
Hi,
see attached file.
I did not try to call to esp32....what is the sip number from esp32 voip ?

songshan99, 2019-09-01:
The username you set in esp-adf is the number to call.

Patsjemoe, 2019-09-01:
and what is the sip sever/domain name ? is this just the local ip address ?

songshan99, 2019-09-01:
The address of your freepbx server.

Patsjemoe, 2019-09-02:
Hi,
I am sorry, but start to be puzzled....where do I find the username in
esp-adf ?

Patsjemoe, 2019-09-02:
with the freepbx as ip address, I get a 500 internal server error.
when using my espVOIP board address, I see
invite sip: ....
register sip: ...

Patsjemoe, 2019-09-02:
when using linphone as server, and my esp32 board ip address as domain, I
get following in my ESP32 screen :

INVITE sip:300@192.168.0.228:5062 SIP/2.0
Via: SIP/2.0/UDP 192.168.0.187:56030
;rport;branch=z9hG4bKPj9e0db008b6ac479e9da989ed234d5244
Max-Forwards: 70
From: "esp32" <sip:ludok@192.168.0.228>;tag=45c44cb2b12e450e90bb554434c1daf8
To: <sip:300@192.168.0.228>
Contact: "esp32" <sip:ludok@192.168.0.187:56030;ob>
Call-ID: b0b23e93a54f468099c95f5444eaa98a
CSeq: 7653 INVITE
Allow: PRACK, INVITE, ACK, BYE, CANCEL, UPDATE, INFO, SUBSCRIBE, NOTIFY,
REFER, MESSAGE, OPTIONS
Supported: replaces, 100rel, timer, norefersub
Session-Expires: 1800
Min-SE: 90
User-Agent: MicroSIP/3.19.18
Content-Type: application/sdp
Content-Length:   343

v=0
o=- 3776411931 3776411931 IN IP4 192.168.0.187
s=pjmedia
b=AS:84
t=0 0
a=X-nat:0
m=audio 4000 RTP/AVP 8 0 101
c=IN IP4 192.168.0.187
b=TIAS:64000
a=rtcp:4001 IN IP4 192.168.0.187
a=sendrecv
a=rtpmap:8 PCMA/8000
a=rtpmap:0 PCMU/8000
a=rtpmap:101 telephone-event/8000
a=fmtp:101 0-16
a=ssrc:1511420736 cname:05976c5073ca10d8

W (900173) SIP: [1970-01-01/00:14:29]<<======================
E (900183) SIP: Error register device
W (900183) SIP: CHANGE STATE FROM 1, TO 0, :func: sip_register:1334
W (960193) SIP: CHANGE STATE FROM 0, TO 0, :func:

Patsjemoe, 2019-09-02:
Hi,
just for your information, as it may be confusing :
if you see : From: "esp32".....this is coming from microsip , as name used
to whom to send...

Patsjemoe, 2019-09-03:
hi,
is there a way to give the esp board a specific sip account name like
sip:user@domain:port ?

Patsjemoe, 2019-09-04:
Hi,
please confirm

songshan99, 2019-09-04:
That user name and password are supposed to serve this purpose. You set up
a dedicated account in freepbx and put the username password information
here.

Patsjemoe, 2019-09-04:
Hi,
and put the username password information here.....what is "here", ESP32,
microsip...?

songshan99, 2019-09-04:
The password and username you write in esp menuconfig, shall include the
same password and username set in freepbx

Patsjemoe, 2019-09-04:
Hi,
Do you mean username and psw when logging in in freepbx,
or the username and secret when you set up an extension in freepbx ?

songshan99, 2019-09-04:
To set up a extension

Patsjemoe, 2019-09-05:
Hi,
I am looking a bit more in-depth on the sip protocol to understand better.
what I noticed is that it looks as I get registered at linphone(see below)
, but in the ESP screen, I get following :

W (73263) SIP: [1970-01-01/00:01:06]=======================>>
W (77273) *SIP: Read again*
W (81273) SIP: Error read or timeout
E (81273) SIP: Error register device
W (81273) SIP: CHANGE STATE FROM 1, TO 0, :func: sip_register:1334

But when I look into the SIP official document:
https://tools.ietf.org/html/rfc3261#page-42
then I do not find the* SIP: Read again*.
This seems to be something in esp-adf voip SW. Can you confirm where this
can be found/in which file, and what it does ?

note : registration :
[image: image.png]
regards
Ludo

Patsjemoe, 2019-09-05:
I (4173) VOIP_EXAMPLE: Speaker has been created
W (8013) SIP: Read again
W (12013) SIP: Error read or timeout
E (12013) SIP: Error register device

I can find speaker has been created as last line in the VOIP example, but
not SIP:read again... not in the application, not in the sip specification.

Patsjemoe, 2019-09-06:
Hi,
Do I need  to use ESP-ADF Pre-release v2.0-beta1 to be able to get freepbx
to work ? I saw that there is specific support for it.
If so, which tools will have to be installwd together to not have missing
files ?
Regards

Patsjemoe, 2019-09-06:
this is in the doc from the beta1 : 
Audio Protocol
Support 2.0 SIP protocol
Support 2.0 RTP protocol
Support RFC4566-based SDP
Support servers: FreePBX, Yate Server and Freeswitch

How can we access this freepbx support ?