espressif/esp-adf #209: VoIP/SIP 481 Call Leg/Transaction Does Not Exist

nandibrenna, 2019-07-03:
Hi,
I'm trying to get the VoIP example to work, but have the following problem. The ESP32 registers with the server (AVM FritzBox) and can also be called.
But as soon as I try to start a call, I get the message 481 Call Leg/Transaction Does Not Exist.
Could it be that it is the tag that is not set in an ACK?

I tried another sip client (https://github.com/chrta/sip_call) and it works fine. Here the tag is set at ACK.

Log from esp-adf/examples/advanced_examples/voip/:

INVITE sip:100@192.168.1.2:5060 SIP/2.0
Via: SIP/2.0/UDP 192.168.1.55:5060;branch=z9hG4bK-299725565;rport
From: <sip:sipesp32@192.168.1.2:5060>;tag=-1995060286
To: <sip:100@192.168.1.2:5060>
Contact: <sip:sipesp32@192.168.1.55:5060>
Max-Forwards: 70
Call-ID: CB28510AE640450FB999CB2E43F0694E4ABBB086D075
CSeq: 12 INVITE
Expires: 600
User-Agent: ESP32 SIP/2.0
Content-Length: 161
Allow: INVITE, ACK, CANCEL, BYE, NOTIFY, REFER, MESSAGE, OPTIONS, INFO, SUBSCRIBE
Content-Type: application/sdp
Proxy-Authorization: Digest username="sipesp32", realm="fritz.box", nonce="D7A3D258C34130BB", uri="sip:192.168.1.2:5060", response="09e46d316100023653f11085a5ce3b45", algorithm=MD5, nc=00000001, cnonce="f7969326deb7beba"

v=0
o=Z 0 3 IN IP4 192.168.1.55
s=SIP Call
c=IN IP4 192.168.1.55
t=0 0
m=audio 9999 RTP/AVP 8 101
a=rtpmap:8 PCMA/8000
a=rtpmap:101 telephone-event/8000

W (775293) SIP: [1970-01-01/00:12:27]=======================>>
W (775913) SIP: [1970-01-01/00:12:27]<<=====READ 0391 bytes==
I (775923) SIP: 

SIP/2.0 401 Unauthorized
Via: SIP/2.0/UDP 192.168.1.55:5060;branch=z9hG4bK-299725565;rport=5060
From: <sip:sipesp32@192.168.1.2:5060>;tag=-1995060286
To: <sip:100@192.168.1.2:5060>;tag=2E15C07DA9016A35
Call-ID: CB28510AE640450FB999CB2E43F0694E4ABBB086D075
CSeq: 12 INVITE
WWW-Authenticate: Digest realm="fritz.box", nonce="76CAA67F9F3F48AB"
User-Agent: FRITZ!OS
Content-Length: 0

W (775953) SIP: [1970-01-01/00:12:27]<<======================
I (775963) SIP: last cseq = 12
W (775963) SIP: status=401 auth_header=3ffdec20
W (776843) SIP: [1970-01-01/00:12:28]<<=====READ 0391 bytes==
I (776843) SIP: 

SIP/2.0 401 Unauthorized
Via: SIP/2.0/UDP 192.168.1.55:5060;branch=z9hG4bK-299725565;rport=5060
From: <sip:sipesp32@192.168.1.2:5060>;tag=-1995060286
To: <sip:100@192.168.1.2:5060>;tag=2E15C07DA9016A35
Call-ID: CB28510AE640450FB999CB2E43F0694E4ABBB086D075
CSeq: 12 INVITE
WWW-Authenticate: Digest realm="fritz.box", nonce="76CAA67F9F3F48AB"
User-Agent: FRITZ!OS
Content-Length: 0

W (776883) SIP: [1970-01-01/00:12:28]<<======================
I (776883) SIP: last cseq = 12
E (776893) SIP: RECEIVE ERROR status_code 401 
W (776893) SIP: CHANGE STATE FROM 64, TO 4, :func: _sip_uac_process_res:1142
I (776903) VOIP_EXAMPLE: SIP_EVENT_BUSY
I (776913) HTTP_AUTH: Digest sipesp32 fritz.box sipesp33 ha1:651aedc888d8531dc340b1e84fe99eaf

W (776913) SIP: [1970-01-01/00:12:28]=======WRITE 0605 bytes>>
I (776923) SIP: 

ACK sip:100@192.168.1.2:5060 SIP/2.0
Via: SIP/2.0/UDP 192.168.1.55:5060;branch=z9hG4bK-919866492;rport
From: <sip:sipesp32@192.168.1.2:5060>;tag=-152085072
To: <sip:100@192.168.1.2:5060>				
Contact: <sip:sipesp32@192.168.1.55:5060>
Max-Forwards: 70
Call-ID: CB28510AE640450FB999CB2E43F0694E4ABBB086D075
CSeq: 12 ACK
Expires: 600
User-Agent: ESP32 SIP/2.0
Content-Length: 0
Proxy-Authorization: Digest username="sipesp32", realm="fritz.box", nonce="76CAA67F9F3F48AB", uri="sip:192.168.1.2:5060", response="6661671a1cf0764321d5fbc273b472b6", algorithm=MD5, nc=00000001, cnonce="60f3471be2356ea6"

W (776983) SIP: [1970-01-01/00:12:28]=======================>>
W (777003) SIP: [1970-01-01/00:12:28]<<=====READ 0319 bytes==
I (777003) SIP: 

SIP/2.0 481 Call Leg/Transaction Does Not Exist
Via: SIP/2.0/UDP 192.168.1.55:5060;branch=z9hG4bK-919866492;rport=5060
From: <sip:sipesp32@192.168.1.2:5060>;tag=-152085072
To: <sip:100@192.168.1.2:5060>
Call-ID: CB28510AE640450FB999CB2E43F0694E4ABBB086D075
CSeq: 12 ACK
User-Agent: FRITZ!OS
Content-Length: 0

ahhfzhang, 2019-07-04:
@nandibrenna  Thanks for your report.

I found the answer on RFC3261: 

> Once credentials have been located, any UA that wishes to
authenticate itself with a UAS or registrar -- usually, but not
necessarily, after receiving a 401 (Unauthorized) response -- MAY do
so by including an Authorization header field with the request.  The
Authorization field value consists of credentials containing the
authentication information of the UA for the realm of the resource
being requested as well as parameters required in support of
authentication and replay protection.

It seems that we should re-send the 'INVITE' request with the 'Authorization' field.

nandibrenna, 2019-07-04:
That could be the solution. 
Here is a log where after '401 Unauthorized' first an 'ACK' is sent and then an 'INVITE' request with the field Authorization is sent again.

INVITE sip:101@192.168.1.2 SIP/2.0
CSeq: 1481765935 INVITE
Call-ID: 1819580481@192.168.1.55
Max-Forwards: 70
User-Agent: sip-client/0.0.1
From: "TK" <sip:sipesp32@192.168.1.2>;tag=1270216262
Via: SIP/2.0/UDP 192.168.1.55:5060;branch=z9hG4bK-1191391529;rport
To: <sip:101@192.168.1.2>
Contact: "sipesp32" <sip:sipesp32@192.168.1.55:5060;transport=udp>
Content-Type: application/sdp
Allow: INVITE, ACK, CANCEL, OPTIONS, BYE, REFER, NOTIFY, MESSAGE, SUBSCRIBE, INFO
Content-Length: 211

v=0
o=sipesp32 812669700 812669700 IN IP4 192.168.1.55
s=sip-client/0.0.1
c=IN IP4 192.168.1.55
t=0 0
m=audio 7078 RTP/AVP 8 101
a=sendrecv
a=rtpmap:101 telephone-event/8000
a=fmtp:101 0-15
a=ptime:20

---------------------------------------------------------------------------

SIP/2.0 401 Unauthorized
Via: SIP/2.0/UDP 192.168.1.55:5060;branch=z9hG4bK-1191391529;rport=5060
From: "TK" <sip:sipesp32@192.168.1.2>;tag=1270216262
To: <sip:101@192.168.1.2>;tag=F32FD1F7B4408171
Call-ID: 1819580481@192.168.1.55
CSeq: 1481765935 INVITE
WWW-Authenticate: Digest realm="fritz.box", nonce="39DBF4723FD84C6E"
User-Agent: FRITZ!OS
Content-Length: 0

---------------------------------------------------------------------------

ACK sip:101@192.168.1.2 SIP/2.0
CSeq: 1481765935 ACK
Call-ID: 1819580481@192.168.1.55
Max-Forwards: 70
User-Agent: sip-client/0.0.1
From: "sipesp32" <sip:sipesp32@192.168.1.2>;tag=1270216262
Via: SIP/2.0/UDP 192.168.1.55:5060;branch=z9hG4bK-1191391529;rport
To: <sip:101@192.168.1.2>;tag=F32FD1F7B4408171
Content-Length: 0

---------------------------------------------------------------------------

INVITE sip:101@192.168.1.2 SIP/2.0
CSeq: 1481765936 INVITE
Call-ID: 1819580481@192.168.1.55
Max-Forwards: 70
User-Agent: sip-client/0.0.1
From: "TK" <sip:sipesp32@192.168.1.2>;tag=1270216262
Via: SIP/2.0/UDP 192.168.1.55:5060;branch=z9hG4bK-553475508;rport
To: <sip:101@192.168.1.2>
Contact: "sipesp32" <sip:sipesp32@192.168.1.55:5060;transport=udp>
Authorization: Digest username="sipesp32", realm="fritz.box", nonce="39DBF4723FD84C6E", uri="sip:101@192.168.1.2", response="57d6b9dee28aaca7fde15251289a1cc5"
Content-Type: application/sdp
Allow: INVITE, ACK, CANCEL, OPTIONS, BYE, REFER, NOTIFY, MESSAGE, SUBSCRIBE, INFO
Content-Length: 211

v=0
o=sipesp32 812669700 812669700 IN IP4 192.168.1.55
s=sip-client/0.0.1
c=IN IP4 192.168.1.55
t=0 0
m=audio 7078 RTP/AVP 8 101
a=sendrecv
a=rtpmap:101 telephone-event/8000
a=fmtp:101 0-15
a=ptime:20

---------------------------------------------------------------------------

SIP/2.0 100 Trying
Via: SIP/2.0/UDP 192.168.1.55:5060;branch=z9hG4bK-553475508;rport=5060
From: "TK" <sip:sipesp32@192.168.1.2>;tag=1270216262
To: <sip:101@192.168.1.2>
Call-ID: 1819580481@192.168.1.55
CSeq: 1481765936 INVITE
User-Agent: AVM FRITZ!Box 7590 154.07.01 (Aug 29 2018)
Content-Length: 0

nandibrenna, 2019-08-11:
Is there any chance to get this issue fixed? Or to get the source of libesp_sip?

ahhfzhang, 2019-08-12:
yes, it's has been fixed, we will update soon.

closed 2019-08-20 by igrr

sosaucily, 2019-10-28:
I'm also interested in seeing the source code for the esp_sip library. Could help extend the functionality to new codecs and such.