espressif/esp-adf #211: Unable to use GPIO peripherals.

LocoMatt, 2019-07-09:
I'm trying to implement external touchpad module (TTP229) into my projec. As I noticed GPIO0 and GPIO2 are unused so I started playing with them. When I test my code "standalone", without any ADF elements in my code it works very well. Below is code which drives keypad. 

```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "sdkconfig.h"

#define SCL_PIN GPIO_NUM_2
#define SDO_PIN GPIO_NUM_0

int pad_num;
int prev_state = 0;
bool is_key_pressed = false;
int x; //store last button state to chec change in read_pad_state()

//return value of pressed button
int get_pad_info(void)
{
  int pad_state = 0;

  for (int i = 1; i <= 16; i++)
  {
    gpio_set_level(SCL_PIN, 0);

    if (!gpio_get_level(SDO_PIN))
    {
      pad_state = i;
    }
    gpio_set_level(SCL_PIN, 1);
    // vTaskDelay(1/ portTICK_PERIOD_MS);
  }
  return pad_state;
}

//return button state
void read_pad_state()
{
  pad_num = get_pad_info(); 

  if (pad_num != prev_state)
  {
    if (pad_num != 0)
    {
      printf("Pad num: %d is pressed \n", pad_num);
      is_key_pressed = true;
      x = pad_num;
    }

    if (pad_num == 0)
    {
      printf("Pad num: %d is released \n", x);
      is_key_pressed = false;
    }
    prev_state = pad_num;
  }
  vTaskDelay(10 / portTICK_PERIOD_MS);
}

void app_main()
{

  gpio_set_direction(SCL_PIN, GPIO_MODE_OUTPUT);
  gpio_set_direction(SDO_PIN, GPIO_MODE_INPUT);
  gpio_set_direction(GPIO_NUM_22, GPIO_MODE_OUTPUT);

  while (1)
  {

    read_pad_state();

    //turn LED on
    if (is_key_pressed == true)
    {
      gpio_set_level(GPIO_NUM_22, 1);
    }
    //turn LED off
    if (is_key_pressed != true)
    {
      gpio_set_level(GPIO_NUM_22, 0);
    }

  }
}

```

When I try to add it into ADF example it doesn't work/behaves very randomly, pressing buttons on board (REC/MODE/BOOT) triggers this part of code and returns random pads pressed. 
I started digging it and I found out that it happens after starting SD card peripheral (when I run this before starting it works well).  
`esp_periph_start(set, sdcard_handle);`
Althought I'm able to turn on/off LED. It's quite strage because almost whole part of this code is just turning on/off GPIO and reading state. I suppose it might be some issues with interrupts but I'm unsure. Hope you can help me!

LocoMatt, 2019-07-09:
Noticed that GPIO0 is used by codec chip. Changed GPIO defines to GPIO4 and GPIO12, also used separated task to drive keypad, seems that everything works fine now.

stasartcode, 2019-07-09:
Audio project without SD/MMC card? GPIO4 and GPIO12 are used by SD card (slot TF_CARD). GPIO4 ->  SD_HOST_D1, GPIO12 ->SD_HOST_D2. Without SD it should work ...

LocoMatt, 2019-07-10:
SD card is used in one line mode so GPIO4 and GPIO12 are free. Now I noticed that 
`esp_err_t ret = audio_event_iface_listen(evt, &msg, portMAX_DELAY);`
seems to be blocking function but without it watchdog is triggered. In while loop I have recording function triggered by REC button (like shown in examples) 
```
   if ((int)msg.data == get_input_rec_id())
         {       
             if (msg.cmd == PERIPH_BUTTON_PRESSED)
             {
               
                 audio_pipeline_stop(pipeline_play);
                 audio_pipeline_wait_for_stop(pipeline_play);
                 ESP_LOGE(TAG, "Recording..."); //loge inny kolor
                 i2s_stream_set_clk(i2s_writer_el, SAMPLE_RATE, BITS, CHANNEL);
                 audio_element_set_uri(fatfs_writer_el, file_path[counter]); 
                 audio_pipeline_run(pipeline_rec);
             }
             else if (msg.cmd == PERIPH_BUTTON_RELEASE || msg.cmd == PERIPH_BUTTON_LONG_RELEASE)
             {
                 audio_pipeline_stop(pipeline_rec);
                audio_pipeline_wait_for_stop(pipeline_rec);
             }
       }
```

and it works well but when I want to add anything else in this function (like turning on and off LED outside this If statement) I can't.

jason-mao, 2019-08-17:
@LocoMatt Is there anything else?  I am not sure I got your exactly means.

LocoMatt, 2019-08-17:
@jason-mao 
I'll describe simple example. 

I have code that uses "REC" button using
`#include "periph_button.h"`

So code looks like that - if REC button is pressed audio is recorded, when released - recording is finished and pipeline stopped 
```
 if ((int)msg.data == get_input_rec_id())
         {       
             if (msg.cmd == PERIPH_BUTTON_PRESSED)
             {
                 ESP_LOGE(TAG, "Recording..."); //loge inny kolor
                 i2s_stream_set_clk(i2s_writer_el, SAMPLE_RATE, BITS, CHANNEL);
                 audio_element_set_uri(fatfs_writer_el, file_path[counter]); 
                 audio_pipeline_run(pipeline_rec);
             }
             else if (msg.cmd == PERIPH_BUTTON_RELEASE || msg.cmd == PERIPH_BUTTON_LONG_RELEASE)
             {
                 audio_pipeline_stop(pipeline_rec);
                audio_pipeline_wait_for_stop(pipeline_rec);
             }
       }
```
This code mentioned above is inside 
`while()`
So it is executed in loop.
To be able to execute this part of code we need to listen to the events
`esp_err_t ret = audio_event_iface_listen(evt, &msg, portMAX_DELAY);`

Now I want to add blinking led in my code. So I'm gonna set my GPIO high and low every bits of time.
```
//count some time
 gpio_set_level(LED, 1);
//count some time
 gpio_set_level(LED, 0);
```
And it just won't work, basically anything I'd like to do with GPIO in IDF - way doesn't work in same while() loop where 
`esp_err_t ret = audio_event_iface_listen(evt, &msg, portMAX_DELAY);`
is placed. 
I have to do it in separated task or write whole peripherals functions from scratch.

In TLDR
In my opinion 
`esp_err_t ret = audio_event_iface_listen(evt, &msg, portMAX_DELAY);` 
is function which blocks code and you're unable to do anything (in same task) that's not related to ADF.

ninghuiy, 2019-10-02:
@LocoMatt   You can try to change "portMAX_delay" to 0

closed 2022-02-08 by jason-mao