espressif/esp-adf #221: SIP example free online sip server like sip2sip

wimmmb, 2019-07-20:
I only get voip working with FreePBX, it seems not possible to connect to a online server /service or other sip servers.
Another example would be appreciated connection to a free online service like sip2sip
Thanks

ahhfzhang, 2019-08-13:
Thanks for your report and i have registered an account in sip2sip, it seems that proxy server needs to be set separately,but i haven't found the relevant standard in RFC3261 yet. 
anyway,we will definitely support some free online sip servers in the future.

Patsjemoe, 2019-08-26:
Dear wimmmmb : can you please confirm all your freepbx settings, because I tried about everything and still get following errors : 

(765963) SIP: Conecting...
W (765963) SIP: CHANGE STATE FROM 0, TO 1, :func: sip_connect:1319
W (765963) SIP: [1970-01-01/00:12:22]=======WRITE 0600 bytes>>
I (765973) SIP:
.....
......then :
W (766023) SIP: [1970-01-01/00:12:22]=======================>>
W (770033) SIP: Read again
W (774033) SIP: Error read or timeout
E (774033) SIP: Error register device
W (774033) SIP: CHANGE STATE FROM 1, TO 0, :func: sip_register:1334

Patsjemoe, 2019-08-28:
Dear wimmmmb : can you please confirm your freepbx settings..you seem to be the only one confirming it worked...

wimmmb, 2019-08-30:
Hi all, I did get the example working, I need to check the settings, think the basic ones as in the example worked. The problem was the (free) services on the internet like sip2sip did not work. I think I had it working with microSIP and/or Officesip with basic settings

Patsjemoe, 2019-08-31:
Hi,
Thanks.
Do you still have the microsip and freepbx settings ?

closed 2022-02-08 by jason-mao