espressif/esp-adf #223: ESP-ADF examples failing to compile

lachlanleslie1995, 2019-07-23:
Hello espressif team,

I was looking into using the esp audio framework and experience a large number of errors that I was unable to results examples such as:
C:/esp-adf/components/audio_hal/driver/zl38063/example_apps/tw_spi_access.c:253:9: error: 'memset' used with length equal to number of elements without multiplication by element size [-Werror=memset-elt-size]
         memset(val, 0, 128);
         ^~~~~~
But after fixing all the memset issues in the components I ran into a bunch of issues with the wifi drivers with similar errors that I basically attributed to changes in naming schemes where structures were unrecognised e.g.
C:/esp-adf/components/esp_peripherals/periph_wifi.c:102:46: error: unknown type name 'smartconfig_status_t'; did you mean 'smartconfig_type_t'?
 static void _wifi_smartconfig_event_callback(smartconfig_status_t status, void *pdata)
                                              ^~~~~~~~~~~~~~~~~~~~
                                              smartconfig_type_t

This was when I attempted to run any of the examples a tried a variety from several sections

Please advise on the best solution or when the framework will be fixed and be usable

vikramdattu, 2019-07-28:
What's the IDF version are you using?
Please use release/v3.2

patrickomatik, 2019-07-28:
I *think* the fixes for this are:

Lines 168, 169 of  ~/esp/esp-adf/components/audio_hal/driver/zl38063/example_apps/tw_hal_verify.c

`    memset(val, 0, MAX_WORDS_FOR_MULTIWORD_ACCESS_TEST * sizeof(short int));
    memset(tempbuf, 0, MAX_WORDS_FOR_MULTIWORD_ACCESS_TEST *sizeof(short int));
`

Line 253 of 
~/esp/esp-adf/components/audio_hal/driver/zl38063/example_apps/tw_spi_access.c

`        memset(val, 0, 128 * sizeof(unsigned short));
`

And line 206 of 
~/ esp/esp-adf/components/esp-adf-libs/esp_codec/downmix.c
`                memset(downmix->inbuf0, 0, bytes_one*downmix->downmix_info.channel[0]);
`

(I'm not 100% sure this last fix to downmix is correct)

But once these patches are applied, there are then further issues which I can't fix (see AUD-629)

:-(

patrickomatik, 2019-07-28:
These issues are with IDF SDK branch v3.2.2

jason-mao, 2019-08-13:
@patrickomatik Thanks for your report. Could you tell me which toolchain version you used?

closed 2019-08-14 by lachlanleslie1995