espressif/esp-adf #225: Example pipeline_passthru doesn't work

api-monitor, 2019-07-24:
I can't hear the audio on AUX_IN with my headphones.

Version of sdk: v2.0 beta1
Device: ESP32_LyraT v4.3
Code: example pipeline_passthru (configured in menuconfig with ESP32_LyraT v4.3)

Configuration:
PC audio port (or smathphone) connect to AUX_IN
Headphones jack to PHONE JACK 
I know for sure that audio cable and headphones are good, as also the port PHONE JACK and audio PC jack.

withdra, 2019-08-14:
@api-monitor This is our mistake, please modify line 30 to     

> audio_hal_ctrl_codec(board_handle->audio_hal, AUDIO_HAL_CODEC_MODE_LINE_IN, AUDIO_HAL_CTRL_START);

and then it will works well.

api-monitor, 2019-08-14:
Thanks
Now it works very well, but I have also add "audio_hal_set_volume(board_handle-> audio_hal, 100)" before start the pipeline. 
I'm not sure but in the first test I heared very low.

closed 2019-08-17 by igrr