espressif/esp-adf #226: Can't compile ADF example programs such as PLAY_MP# because of errors

jponko, 2019-07-25:
Just got my lyraT board today but I can't compile the ADF example programs because of numerous errors such as the one below for touch pads. Is there something else I should select in menuconfig besides just selecting  the ESP32-lyraT V4.3 board in Audio HAL?

ESP_IDF sample programs compile and run.  
 
/esp/esp-adf/components/audio_board/lyrat_v4_3/board_def.h:34:35: error: 'TOUCH_PAD_NUM8' undeclared (first use in this function)
 #define BUTTON_PLAY_ID            TOUCH_PAD_NUM8

patrickomatik, 2019-07-28:
I have the above issue too. In addition, if compiling for v2.2 boards, I got various memset errors - current versions of the compiler require the size of the type being copied to explicitly included. I patched these, only to then encounter issues due to non-existence of gpio_uninstall_isr_service() services (and the online doxygen API docs for gpio_uninstall_isr_service are also broken).

It does seem like someone from Espressif either needs to review the ADF examples for each board type and check they build against the latest stable IDF/compiler combos, because they are definitely completely broken at the moment, or provide clearer guidance on which specific IDF version should be used - 'latest stable' is insufficient.

vikramdattu, 2019-07-28:
Please use IDF version: 'release/v3.2'

patrickomatik, 2019-07-28:
My issues described above are with branch v3.2.2

jponko, 2019-07-28:
I was able to fix my problem by installing IDF v3.2.2. Thanks
jp AC9NM

closed 2019-07-28 by jponko