espressif/esp-adf #230: esp-adf in arduino

Patsjemoe, 2019-08-09:
it would be goid if esp-adf or the audio kit/lyraT could be used in arduino, because the mingw32 environment is completely useless. All internet suggestions for this do not work properly.

timotoots, 2019-08-11:
Yes, ADF in arduino would be great indeed!

Andre2020, 2019-08-13:
I don't think the mingw32 environment is completely useless. I have worked on mingw32 before. If you think mingw32 make is slow. I suggest you use WSL on windows or MAC/Linux.

BTW，I am also wonder esp-adf-arduino when will be relesed. Does the development team have a schedule on it?

Patsjemoe, 2019-08-13:
Hi,
It is not just slow, I could live with that , I cannot get it to work. Make
does not work, whatever paths I use. It always shows errors.
I am now trying a clean install, only focussing on cmd. Make works here
now, but I get errors when compiling the play_mp3.
Tbc....to be continued
Regards
Ludo