espressif/esp-adf #231: ESP making CON file in windows that is not allowed in windows

Patsjemoe, 2019-08-11:
When I compile projects in BLINK or HELLO_world, the Espressif SW make a file called "CON" in the directories ;
blink/build/include/config/smp/slave/con
hello_world/build/include/config/smp/slave/con
When trying to delete these files, windows says this is not accessible, entrance not valid, location not available.
"con" is a windows reserved name that is not allowed to use, so I wonder if this is creating my problems with MING32 path's and errors.

Alvin1Zhang, 2019-08-13:
Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Thanks. @Patsjemoe

Patsjemoe, 2019-08-16:
I have done new installations with esp-idf v3.2.2, without installing automatic toolchain 2.0 (which seems not to be allowed with v3.2.2......) now, the con problem is no longer showing.

closed 2019-08-16 by Patsjemoe